vroland/epdiy #91: Question about epd_task in weather example

smartinick, 2021-06-25:
Hi there!

i just "ported" the weather example to platform.io, where, in the end it's just to rename weather.cpp to main.cpp, add a platformio.ini and.... fix the "app_main" error which will be defined 2 times now.
when fixing the app_main - moved the content to setup(), i noticed the epd_task, and i'm wondering why it's there...?
it's just doing some calls before looping endlessly.

for testing i removed that task, commented the xTaskCreatePinnedToCore(&epd_task... line and copied the lines from epd_task to setup(). a first test shows me it's working....

and now i'm wondering why there's this extra task that's just looping.....

and No, that shall not be critics, i like to learn & understand!
i'm also fine if it's just the way it is without any further reason behind it....

Thanks & Brgds!,  Martin.

martinberlin, 2021-06-26:
I also don’t know why it’s there, but it just initializes the EPD driver and sets the memory buffer to white. I guess it was like this in the original program and @vroland ported it like this.
It really does not matter that this loops and stays there since the rest of the program keeps doing what it intends to do. But yes I guess it can be moved out into main or setup and it will work the same as long as the epd_init and frame buffer initialization happens before any other epaper communication.

smartinick, 2021-06-27:
Great, Thanks for the feedback!

i though maybe there's some reason like having parallel init prozess to save processing time... :)
currently i'm putting all that code into a class, adding some other classes for touch/mqtt-dashboad and there might be quite many screen updates.... so i thought about utilizing a separate task for that, using rtos queues to queue the partial-screenupdates and have them process asap in the background.... and there's the parallization idea again ;)

Thanks!

vroland, 2021-07-07:
Are you using the esp-idf or the arduino framework with platformio? If I got your question correctly, then app_main is the entry point for the esp-idf and setup() and loop() for the arduino framework. The weather example specifically was originally written for arduino, then ported to the IDF (and now maybe back to arduino ;) ), so there might be some inconsistencies in the idioms.
So if it works for you, you can just simplify it :)

mcer12, 2022-04-22:
Closing, stale.

closed 2022-04-22 by mcer12