vroland/epdiy #92: Lilygo T5-4.7 screen turns darker grey while on battery

rmk92, 2021-07-10:
I've just connected a battery to my Lilygo T5, and I notice that as soon as the power is cut (blue LED turns off) the white areas of the display starts turning darker. I'm using epd_power_on() and epd_power_off() to just power the display during updates. This doesn't happen when connected to USB, when the hardware ensures that the display is cosntantly powered.

I haven't investigated with a 'scope yet, but I suspect the issue may be that the e-ink display isn't being allowed to properly power down. Does epdiy guarantee that the signals to the e-ink display get set to logic low or outputs disabled while the power to the e-ink display is turned off?

There are other issues raised with this which suggest people eventually see their display failing - this might be explained if the display is remaining semi-powered through the logic signals? Please remember that it is quite normal to have diode clamps to the power rails on silicon devices.

vroland, 2021-07-11:
Interesting theory, I didn't really consider that it until now. I mainly used the e-reader screens and I never had an issue with them.
Could you try calling `epd_deinit()` after turning off the power to the display? Maybe you need to additionally set the data pins as inputs to isolate them after that. If that changes anything, we should probably add that to the driver.

mcer12, 2021-07-15:
@rmk92 @vroland  powering the display with signals is not the case, those go to the display controller which then controls mosfets with high voltages to change the pixels. You can just cut the high voltage rails and see what the display does without them :D 

What I suspect is that when you just cut power, the display power rails are not shut down in correct order and that's why you see this phenomenon. Bottom line is - for the display to work correctly, you need to use deep sleep. Your other options would be to sense power outage, and immediately run the display shutdown and deinit functions (you probably won't have time for that before the capacitors drain). Or simply shutdown & deinit before you disconnect power - with a button or whatever.

lixy123, 2022-02-08:
as  the code of "https://github.com/vroland/epdiy/tree/master/examples/lilygo-t5-47-epd-platformio",
if you use battery power and not usb power. When the ink screen enters the deep_sleep after call "esp_deep_sleep_start", a large number of black spots will be added at the white display of the display screen, as shown in the following figure

before esp_deep_sleep_start:
[image]

after esp_deep_sleep_start:
[image]

I compared the official driver code of LilyGo-EPD47 and epdiy,
"https://github.com/vroland/epdiy/blob/master/src/epd_driver/display_ops.c"  
and
"https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/src/ed097oc4.c"
add the following functions to "display_ops.c" and "display_ops.h" 

》》》》》
void epd_poweroff_all();  
void epd_poweroff_all()
{
    /*only for: LILYGO_T5_47
    call the function before call esp_deep_sleep_start(),
    otherwise, There will be a lot of clutter after LILYGO ink screen go to  sleep.
    */
    #if defined(CONFIG_EPD_BOARD_REVISION_LILYGO_T5_47)
    memset(&config_reg, 0, sizeof(config_reg));
    push_cfg(&config_reg);
    #endif
}
》》》》》

Above , maybe  can solve the display problem you encounter,  The screen should not appear a large number of dark spots after hibernation on lilygo display screen. 
Although it's not good to modify the epdiy source code without reading through other people's code, 
but it's useful

vroland, 2022-02-09:
You're right, all pins of the config register should be set to low on poweroff(). Actually, this is done for V4, V5 and V6, but the LILYGO board uses the config_reg_v2.h header, where this is not implemented. Would you mind submitting a pull request adding it to the config_reg_v2.h header, like in config_reg_v4.h?

lixy123, 2022-02-17:
I don't mind. You're a code expert. I'm not familiar with the code modification of partial hardware. In fact, I'm just a web designer.
I changed it very unprofessional and the test is feasible only.

martinberlin, 2022-02-17:
Thanks for the suggestion. I can take care of doing this PR so you can assign it to me if you like

xcybplx, 2022-02-17:
I also have the same issue. Is there a way to solve it? I don't use deep sleep in the esphome, just connected battery and the screen go darker and darker.

lixy123, 2022-02-18:
> I also have the same issue. Is there a way to solve it? I don't use deep sleep in the esphome, just connected battery and the screen go darker and darker.

These are two questions :

1. When refreshing the display of new content, after called poweroff() , screen will  darken slowly

2. If you go to sleep, after called goto_sleep(),  the screen background will darker  slowly, and more obvious

I have submited the two issues,  
The corresponding phenomena are described in detail:
https://github.com/vroland/epdiy/issues/92
https://github.com/vroland/epdiy/issues/159

Don't worry, the author will help solve it

mcer12, 2022-04-22:
@martinberlin what's the status with this? Can this be closed?

martinberlin, 2022-04-22:
Yes. I added already the requested poweroff to the lines in #159

The darker effect on the Lilygo is a hardware issue that IMHO should be reported on their own repository. And I experience the same with my own Lilygo test display. Thanks Martin

closed 2022-04-22 by martinberlin