vroland/epdiy #94: Fontconvert with big fonts generate out of range glyph width/height and advance

giotherobot, 2021-07-20:
Hello I'm trying to use the epdiy driver with a lilygo board.
I have issues when trying to use some big fonts, ex 128 or 200, that i need since the display will be used to show some telemetry data in a boat and the font size needs to be large to see it clearly. 
I have found the problem to be that `EpdFont` and `EpdGlyph` structs contains `uint8_t` values. fontconvert.py seems to ignore the fact that `uint8_t` nubers must be less than 256 and writes bigger numbers in the slots, hece we get compilation errors.
Are there any possible solutions or is it a hardware limitation?

vroland, 2021-07-21:
Hi, I think this was just me not thinking about bigger fonts and not a hardware limitation. Actually, changing the type to uint16_t should be fairly seamless. So if you want to give it a try I'm happy to accept a pull request :)
Maybe some functions in font.c need adjusting as well.

giotherobot, 2021-07-21:
We'll give it a try and let you know then!

martinberlin, 2022-05-10:
This fix was already merged, closing here