vroland/epdiy #95: Does Epdiy support ED060XCG,VB3300-FOC?

lanistor, 2021-07-22:
It's really an amazing, aowsom project, support E-Ink screens without TCON IC.
Since we neally cannot bought TCON IC, also it really costs.

This will help us to save a lot of money, and reduce supply chain risks.

### 
Then, Does Epdiy support ED060XCG,VB3300-FOC?

ED060XCG,VB3300-FOC's resolution are all 1024*758 (212), should they support as ED060XC3?

vroland, 2021-07-23:
Hi,
I didn't come across the ED060XCG yet, but if the pinout matches the XC3 it probably works. However, if you don't have the matching waveform files you will have to figure out the timing yourself for accurate grayscale. Do you have a data sheet for the display?

lanistor, 2021-07-26:
> Hi,
> I didn't come across the ED060XCG yet, but if the pinout matches the XC3 it probably works. However, if you don't have the matching waveform files you will have to figure out the timing yourself for accurate grayscale. Do you have a data sheet for the display?

Thanks for attention.
ED060XH7 is same as VB3300-FOC. [ED060XH7](https://raw.githubusercontent.com/lanistor/assets/master/screens/ED060XH7-Ver%201.0-P-511-734-V1.pdf)
[ED060XCG](https://raw.githubusercontent.com/lanistor/assets/master/screens/ED060XCG%20Final%20SPEC_V2_P-511-869-V2_.pdf)

vroland, 2021-07-26:
So far, I have tested the XH3 and SCT. But since the XH7 is has the same pinout as the XH7, i don't see why it won't work. You will however need to calibrate the grayscale timings.

lanistor, 2021-07-30:
> So far, I have tested the XH3 and SCT. But since the XH7 is has the same pinout as the XH7, i don't see why it won't work. You will however need to calibrate the grayscale timings.

We havn't test XH7, it's said has the same waveform with XH7, it should be runnable normally, we are testing it.

lanistor, 2022-01-13:
Supported, had tested, thanks.

closed 2022-01-13 by lanistor