vroland/epdiy #96: Rotation in epaper-breakout-top-jlcpcb.csv maybe not correct

Li-Tianming, 2021-07-26:
In epaper-breakout-top-jlcpcb.csv, [U5 component](https://github.com/vroland/epdiy/blame/master/hardware/epaper-breakout/gerbers/epaper-breakout-top-jlcpcb.csv#L74) rotation is 90 degree, 

but in JLC's component description, it's  original orientation is like below, 

[JLC's STM search system](https://www.jlc.com/portal/smtComponentList.html)

[image]

[image]

If i use 90 as rotation, in JLC_assist PC program preview, it will looks like this, obvious it's not correct.
[image]

so i think the correct rotation maybe is 270 degree

vroland, 2021-07-26:
Hi, so far I always soldered the ESP32 module by hand, so this is probably just a mistake. Unfortunately, rotations as exported by KiCad do not correspond to JLCPCB for some packages. I tried to fix most of them, but the ESP is not a standard package.
If 270° looks better, it's probably correct :) Fortunately, they usually fix such things when it's as obvious as with the ESP.
If you want, feel free to submit a pull request with the fixed rotation.

Li-Tianming, 2021-07-27:
thanks for the reply.  

i am trying to make thing correct and works, if thing goes well, i will create a pull request to update the jlc's SMT file.

2dom, 2021-08-17:
Yeah ... same here.  JLCPCB shows all the parts (even resistors etc) shifted with v5. Not usable for ordering SMT assembly unfortunately :(

vroland, 2021-08-22:
Hm, resistors, capacitors and diodes always where correct for me... That's strange. Is that still the case when you manually export the position file again from KiCad?

Li-Tianming, 2021-09-05:
[image]

everything works pretty fine

mcer12, 2021-11-27:
@Li-Tianming please note that the JLCPCB assembly is kinda broken so it sometimes shows the footprints wrong. They always manually fix it and send you a screenshot so that you can check. Don't worry about it :)