vroland/epdiy #98: U4 vreg unnecessary?

sebastius, 2021-08-07:
[image]

If you hook up pin 16 of U12 (CH340C) to VBUS, and pin 4 to a 100nF cap, then you can drop U4 and C13/C14. Saves you a few cents and 1cm^2 of board space :)

I'm currently building a board but i don't have all the parts to verify. Can confirm that the USB-serial is working.

vroland, 2021-08-09:
Since the CH340C would be in 5V mode, wouldn't the output signals 5V as well? Afaik, although the esp32 seems to be practially 5V tolerant (at least for some time), technically it could burn out the input pins. So I'd rather be on the safe side :)

mcer12, 2021-08-18:
@sebastius  ESP32 is NOT 5V tolerant as opposed to ESP8266. If you look up schematics of the development boards, they all use 3V3 to power the CH340.

vroland, 2021-08-22:
I think this question was answered. If not, feel free to re-open @sebastius :)

closed 2021-08-22 by vroland