vroland/epdiy #100: pi pico support?

deanhuff, 2021-08-20:
Hello, i'm working on a raspberry pi pico based E-ink typewriter.  would the epdiy be something i could use to drive a 6-10" panel with partial updates from a pico?  I have a small 2.9" waveshare display and it is pretty [darn slow](https://www.youtube.com/watch?v=zjWYZ1kRimg&ab_channel=DeanHuffman) with partial updates (as far as typing is concerned).

martinberlin, 2021-08-20:
This is using the FreeRTOS real-time OS and is designed for ESP32 „system on a chip“. I’ve seen that there where some efforts to compile freeRTOS on the raspberry PI but still, this uses ESP-IDF libraries like I2S that won’t be available on the PI. But that’s only my advice since I’m not the maintainer of this repository.
Just wanted to express my sincere opinion.
The best will be to find a project that runs on the raspberry PI directly and it’s focused on driving parallel epapers using the PI gpios.

vroland, 2021-08-22:
@deanhuff Hm, as @martinberlin said, this would be a lot of work. In general, the Pi Pico is a lot slower than the ESP32. Not sure if you could compensate for this with some programmable I/O trickery, but porting won't be straightforwared.
It is probably easier to design some extra hardware that is responsible for driving the display, but at this point there is not really a point in using the Pico any more :D But I'm not familiar enough with it to say it's not possible ;)

deanhuff, 2021-08-22:
Thanks for the response!  I'll continue to research to see what my options are.

martinberlin, 2024-08-19:
Since a few days there is a new [Raspberry PICO2](https://www.raspberrypi.com/products/raspberry-pi-pico-2/) I decided to make a "HAT board" PCB and route signals + Voltages to see if it's possible to drive it:

https://github.com/martinberlin/H-pico-raw-eink

> the RP2350 can control up to 16MB of PSRAM and 16MB of FLASH.

 **I like because is original. No one did it so far!**

I'm doing this with my colleague Larry which has already experience with PICO boards. I lack the right experience to do the software since I only used Linux in the big Raspberry PI. 
Sadly still there is no Arduino IDE support for the new PICO2. All this is really to get deeper into understanding the Eink signals required to drive the display, and add the same time to make an Adapter board that supports:

- EINK voltages using TPS65185 (Just like epdiy version >= 6)
- QWIIC I2C connector
- Small OLED 4 pins connector (Debugging, etc)
- USB-C for power only (PICO still uses micro-USB)'
- MicroSD adapter (If there are enough IOs left)