vroland/epdiy #101: VCOM always 0,38V

M737M737, 2021-08-22:
Hi,

VCOM on my PCB is always 0,38V. I've flashed the esp, but there is no change at the Vcom value when changing the value of the potentiometer.

Also my measurements at the lm358 are not ok:
Pin 8: 4,74V (15V)
Pin 4: 0,38V (-15V)

The 5V and 3,3V pins are ok.

Serial output when booting the esp:
15:08:00.232 -> ets Jun  8 2016 00:22:57
15:08:00.232 -> 
15:08:00.232 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:08:00.232 -> configsip: 0, SPIWP:0xee
15:08:00.232 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:08:00.278 -> mode:DIO, clock div:1
15:08:00.278 -> load:0x3fff0030,len:7348
15:08:00.278 -> load:0x40078000,len:15464
15:08:00.278 -> load:0x40080400,len:4452
15:08:00.278 -> entry 0x40080694
15:08:00.278 -> [0;32mI (27) boot: ESP-IDF v4.3 2nd stage bootloader[0m
15:08:00.278 -> [0;32mI (27) boot: compile time 22:13:25[0m
15:08:00.278 -> [0;32mI (27) boot: chip revision: 1[0m
15:08:00.278 -> [0;32mI (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
15:08:00.278 -> [0;32mI (37) qio_mode: Enabling default flash chip QIO[0m
15:08:00.278 -> [0;32mI (42) boot.esp32: SPI Speed      : 80MHz[0m
15:08:00.325 -> [0;32mI (47) boot.esp32: SPI Mode       : QIO[0m
15:08:00.325 -> [0;32mI (51) boot.esp32: SPI Flash Size : 4MB[0m
15:08:00.325 -> [0;32mI (56) boot: Enabling RNG early entropy source...[0m
15:08:00.325 -> [0;32mI (61) boot: Partition Table:[0m
15:08:00.325 -> [0;32mI (65) boot: ## Label            Usage          Type ST Offset   Length[0m
15:08:00.325 -> [0;32mI (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
15:08:00.325 -> [0;32mI (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
15:08:00.325 -> [0;32mI (87) boot:  2 factory          factory app      00 00 00010000 00100000[0m
15:08:00.325 -> [0;32mI (94) boot: End of partition table[0m
15:08:00.372 -> [0;32mI (99) boot_comm: chip revision: 1, min. application chip revision: 0[0m
15:08:00.372 -> [0;32mI (106) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=09744h ( 38724) map[0m
15:08:00.372 -> [0;32mI (126) esp_image: segment 1: paddr=0001976c vaddr=3ffb0000 size=0295ch ( 10588) load[0m
15:08:00.372 -> [0;32mI (130) esp_image: segment 2: paddr=0001c0d0 vaddr=40080000 size=03f48h ( 16200) load[0m
15:08:00.372 -> [0;32mI (139) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1770ch ( 96012) map[0m
15:08:00.418 -> [0;32mI (171) esp_image: segment 4: paddr=00037734 vaddr=40083f48 size=0b050h ( 45136) load[0m
15:08:00.418 -> [0;32mI (187) esp_image: segment 5: paddr=0004278c vaddr=50000000 size=00010h (    16) load[0m
15:08:00.464 -> [0;32mI (195) boot: Loaded app from partition at offset 0x10000[0m
15:08:00.464 -> [0;32mI (195) boot: Disabling RNG early entropy source...[0m
15:08:00.464 -> [0;32mI (209) psram: This chip is ESP32-D0WD[0m
15:08:00.464 -> [0;32mI (209) spiram: Found 64MBit SPI RAM device[0m
15:08:00.464 -> [0;32mI (209) spiram: SPI RAM mode: flash 80m sram 80m[0m
15:08:00.464 -> [0;32mI (212) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
15:08:00.464 -> [0;32mI (220) cpu_start: Pro cpu up.[0m
15:08:00.464 -> [0;32mI (223) cpu_start: Starting app cpu, entry point is 0x4008120c[0m
15:08:00.464 -> [0;32mI (0) cpu_start: App cpu up.[0m
15:08:00.934 -> [0;32mI (702) spiram: SPI SRAM memory test OK[0m
15:08:00.934 -> [0;32mI (710) cpu_start: Pro cpu start user code[0m
15:08:00.981 -> [0;32mI (710) cpu_start: cpu freq: 240000000[0m
15:08:00.981 -> [0;32mI (710) cpu_start: Application information:[0m
15:08:00.981 -> [0;32mI (713) cpu_start: Project name:     vcom_calibration_helper[0m
15:08:00.981 -> [0;32mI (719) cpu_start: App version:      1.0.0-38-g598f77b[0m
15:08:00.981 -> [0;32mI (725) cpu_start: Compile time:     Jun 30 2021 22:12:57[0m
15:08:00.981 -> [0;32mI (731) cpu_start: ELF file SHA256:  740191c362579d41...[0m
15:08:00.981 -> [0;32mI (737) cpu_start: ESP-IDF:          v4.3[0m
15:08:00.981 -> [0;32mI (742) heap_init: Initializing. RAM available for dynamic allocation:[0m
15:08:01.027 -> [0;32mI (749) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
15:08:01.027 -> [0;32mI (755) heap_init: At 3FFB3318 len 0002CCE8 (179 KiB): DRAM[0m
15:08:01.027 -> [0;32mI (761) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
15:08:01.027 -> [0;32mI (768) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
15:08:01.027 -> [0;32mI (774) heap_init: At 4008EF98 len 00011068 (68 KiB): IRAM[0m
15:08:01.027 -> [0;32mI (780) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
15:08:01.027 -> [0;32mI (788) spi_flash: detected chip: generic[0m
15:08:01.027 -> [0;32mI (793) spi_flash: flash io: qio[0m
15:08:01.073 -> [0;32mI (797) cpu_start: Starting scheduler on PRO CPU.[0m
15:08:01.073 -> [0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
15:08:01.073 -> [0;32mI (813) esp_temperature: Characterized using eFuse Vref
15:08:01.073 -> [0m
15:08:01.073 -> [0;32mI (819) main: waiting for one second before poweron...[0m
15:08:02.056 -> [0;32mI (1825) main: enabling VCOMM...[0m
15:08:02.056 -> [0;32mI (1826) main: VCOMM enabled. You can now adjust the on-board trimmer to the VCOM value specified on the display.[0m

Do you have any idea what the issue could be?

Thanks!

vroland, 2021-08-22:
This sounds like a hardware issue. Somethimes chips are in the wrong orientation. Could you post a high-resolution image of the PCB?

M737M737, 2021-08-22:
[image]

vroland, 2021-08-22:
Hm, nothing looks obviously wrong. Did you select the right board revision in menuconfig? If so, is the 22 / -20V rail working?

M737M737, 2021-08-22:
I'm not sure. Could be the mistake...

4,93V instead of 22V
0V instead of -20V

vroland, 2021-08-22:
Which options do you have selected in menuconfig?

M737M737, 2021-08-22:
I've selected the ED097OC4 - Display (that's the one I ordered) and the v4 Board.
Then I built the configuration helper programm.

Is there anything else to config?

vroland, 2021-08-23:
That seems correct. Do you have any other boards, or just the single one? Maybe you can probe the voltages on the shift register U8 to see if the activation signals for the boosters are enabled?

M737M737, 2021-08-24:
Sorry for the late answer, I had no time for the projekt yesterday...

Excellent news: I retried to calibrated Vcom today and surprisingly it worked.

The problem:
1st try: I flashed without the configuration for the v4 board -> not working
2nd try: Correctly flashed but no additional reset. -> reset the board again before measuring Vcom
3rd try: It worked without any changes

Afterwards I also used the demo program sucessfully! Thanks for your help, I'm glad that it's now running after some months of struggle. 

[image]

vroland, 2021-08-25:
I'm glad to hear that! :) Btw, for cleaner partial updates it is beneficial to add a 4.7uF or 10uF capacitor between VCOM and Ground. Otherwise you tend to see the outlines of partial updates. This is fixed in v6, but not on the older boards by default.

So I guess this issue is solved now? If not, feel free to re-open.

closed 2021-08-25 by vroland