vroland/epdiy #105: fatal error: hal/i2c_types.h: No such file or directory

RealDekkia, 2021-08-28:
Hi,
I tried to use this library with the Lilygo T5.
When compiling it always errors out because it can't find hal/i2c_types.h

When using the official Lilygo-Library (https://github.com/Xinyuan-LilyGO/LilyGo-EPD47) the same thing happens.

Compiling a project that doesn't include epdiy works fine (for the Lilygo as well as other esp32-based stuff)

This is the full log:

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\dekkia\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\dekkia\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\dekkia\Documents\Arduino\libraries -fqbn=esp32:esp32:lilygo_t5_47:DisplayType=default,PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10812 -build-path C:\Users\dekkia\AppData\Local\Temp\arduino_build_906427 -warnings=all -build-cache C:\Users\dekkia\AppData\Local\Temp\arduino_cache_677889 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose C:\Users\dekkia\Documents\Arduino\libraries\epdiy\examples\demo\main\main.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\dekkia\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\dekkia\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\dekkia\Documents\Arduino\libraries -fqbn=esp32:esp32:lilygo_t5_47:DisplayType=default,PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10812 -build-path C:\Users\dekkia\AppData\Local\Temp\arduino_build_906427 -warnings=all -build-cache C:\Users\dekkia\AppData\Local\Temp\arduino_cache_677889 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -verbose C:\Users\dekkia\Documents\Arduino\libraries\epdiy\examples\demo\main\main.ino
Using board 'lilygo_t5_47' from platform in folder: C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
cmd /c if exist "C:\\Users\\dekkia\\Documents\\Arduino\\libraries\\epdiy\\examples\\demo\\main\\partitions.csv" copy /y "C:\\Users\\dekkia\\Documents\\Arduino\\libraries\\epdiy\\examples\\demo\\main\\partitions.csv" "C:\\Users\\dekkia\\AppData\\Local\\Temp\\arduino_build_906427\\partitions.csv"
cmd /c if not exist "C:\\Users\\dekkia\\AppData\\Local\\Temp\\arduino_build_906427\\partitions.csv" copy "C:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "C:\\Users\\dekkia\\AppData\\Local\\Temp\\arduino_build_906427\\partitions.csv"
        1 Datei(en) kopiert.

Detecting libraries used...
"C:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bt" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/coap" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/console" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/driver" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/efuse" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-tls" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp32" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_event" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_client" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_server" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_ota" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_server" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_websocket_client" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/espcoredump" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/ethernet" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/expat" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/fatfs" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freemodbus" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freertos" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/heap" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/idf_test" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/jsmn" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/json" "-IC:\\Us [... cut ...] .c" -o nul
Alternatives for hal/i2c_types.h: []
ResolveLibrary(hal/i2c_types.h)
  -> candidates: []
C:\Users\dekkia\Documents\Arduino\libraries\epdiy\src\epd_driver\pca9555.c:3:27: fatal error: hal/i2c_types.h: No such file or directory

compilation terminated.

Using library epdiy at version 1.0.0 in folder: C:\Users\dekkia\Documents\Arduino\libraries\epdiy 
exit status 1
Error compiling for board LILYGO T5-4.7 inch e-paper.
```

I'm pretty sure I'm doing something wrong, but I'm not sure what.

vroland, 2021-08-29:
Which environment are you using? Arduino IDE? It could be some paths in the new v6 firmware that I have to adjust for the old version in arduino-esp32.

RealDekkia, 2021-08-29:
I'm using the latest Ardino IDE (1.8.15)

vroland, 2021-08-29:
Ok, then, could you try a version before the v6 merge?

RealDekkia, 2021-08-29:
I just tried 1.0.0 from the release-page.
The error doesn't come up there.

It seems that has been the issue.

vroland, 2021-08-29:
The example has to be set up to use setup() and loop() as entry points. I'm not sure if all examples are set up that way, yet. Could you describe the error you are getting?

RealDekkia, 2021-08-29:
Only 'demo' and 'dragon' are set up to work with Arduino. (But only using setup() and loop() as entry-points, but putting the code directly into those funkctions isn't an option?)

'dragon' is the only example that works. When trying to compile 'demo', I get this error:
```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\dekkia\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\dekkia\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\dekkia\Documents\Arduino\libraries -fqbn=esp32:esp32:lilygo_t5_47:DisplayType=default,PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10815 -build-path C:\Users\dekkia\AppData\Local\Temp\arduino_build_795899 -warnings=all -build-cache C:\Users\dekkia\AppData\Local\Temp\arduino_cache_738545 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\dekkia\Documents\Arduino\libraries\epdiy\examples\demo\main\main.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\dekkia\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\dekkia\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\dekkia\Documents\Arduino\libraries -fqbn=esp32:esp32:lilygo_t5_47:DisplayType=default,PartitionScheme=default,FlashMode=qio,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10815 -build-path C:\Users\dekkia\AppData\Local\Temp\arduino_build_795899 -warnings=all -build-cache C:\Users\dekkia\AppData\Local\Temp\arduino_cache_738545 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -verbose C:\Users\dekkia\Documents\Arduino\libraries\epdiy\examples\demo\main\main.ino
Using board 'lilygo_t5_47' from platform in folder: C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: C:\Users\dekkia\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6
cmd /c if exist "C:\\Users\\dekkia\\Documents\\Arduino\\libraries\\epdiy\\examples\\demo\\main\\partitions.csv" copy /y "C:\\Users\\dekkia\\Documents\\Arduino\\libraries\\epdiy\\examples\\demo\\main\\partitions.csv" "C:\\Users\\dekkia\\AppData\\Local\\Temp\\arduino_build_795899\\partitions.csv"
cmd /c if not exist "C:\\Users\\dekkia\\AppData\\Local\\Temp\\arduino_build_795899\\partitions.csv" copy "C:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "C:\\Users\\dekkia\\AppData\\Local\\Temp\\arduino_build_795899\\partitions.csv"
        1 Datei(en) kopiert.
Detecting libraries used...
"C:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" "-IC:\\Users\\dekkia\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/ [... cut ...]
C:\Users\dekkia\Documents\Arduino\libraries\epdiy\examples\demo\main\main.c: In function 'idf_loop':
main.c:98:8: error: implicit declaration of function 'millis' [-Werror=implicit-function-declaration]
   t1 = millis();
        ^
main.c:131:3: error: implicit declaration of function 'delay' [-Werror=implicit-function-declaration]
   delay(1000);
   ^
cc1.exe: some warnings being treated as errors
Using library epdiy at version 1.0.0 in folder: C:\Users\dekkia\Documents\Arduino\libraries\epdiy 
exit status 1
implicit declaration of function 'millis' [-Werror=implicit-function-declaration]
```

vroland, 2021-08-29:
Hm, maybe because arduino.h is missing? Could you try to add #include "Arduino.h" to the main.c of Demo?
Also, you could paste the code directly to the .ino file if you wanted, but I wanted to avoid duplication and for now, the IDF is the primary platform I use.

RealDekkia, 2021-08-29:
Great, thanks! Adding #include "Arduino.h" fixes 'demo'

I now tried to put the code directly into the .ino but it doesn't work.

The 'dragon'-example looks like this now:
dragon.ino:
```c
#include "esp_heap_caps.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "Arduino.h"
#include "epd_driver.h"
#include "epd_highlevel.h"
#include "dragon.h"

EpdiyHighlevelState hl;

extern "C"
{
    void idf_setup();
    void idf_loop();
}

void setup() {
    if(psramInit()){
        Serial.println("\nThe PSRAM is correctly initialized");
    } else{
        Serial.println("\nPSRAM does not work");
    }

    idf_setup();
}

void loop() {
    idf_loop();
}

void idf_loop() {

  EpdRect dragon_area = {
      .x = 0,
      .y = 0,
      .width = dragon_width,
      .height = dragon_height
  };
  epd_copy_to_framebuffer(dragon_area, dragon_data, epd_hl_get_framebuffer(&hl));

  epd_poweron();
  epd_clear();
  int temperature = 25;
  enum EpdDrawError _err = epd_hl_update_screen(&hl, MODE_GC16, temperature);
  epd_poweroff();

  while (1) {
      vTaskDelay(1000);
  }
}

void idf_setup() {
  epd_init(EPD_OPTIONS_DEFAULT);
  hl = epd_hl_init(EPD_BUILTIN_WAVEFORM);
}

#ifndef ARDUINO_ARCH_ESP32
void app_main() {
  idf_setup();

  while (1) {
    idf_loop();
  };
}
#endif
```
dragon.h stayed the same and main.c is empty.

This is the error-message when trying that:

mcer12, 2021-09-02:
@vroland have the same issue in platformio, 1.0.0 release works, V6 release returns the issue. Seems like it's opening V6 specific files even though I have selected V5:

```
Compiling .pio\build\esp32dev\lib1b5\epdiy@src-64f018394500b0723813fd6b74fd5859\epd_driver\tps65185.c.o
Archiving .pio\build\esp32dev\libFrameworkArduinoVariant.a
.pio\libdeps\esp32dev\epdiy@src-64f018394500b0723813fd6b74fd5859\src\epd_driver\pca9555.c:3:27: fatal error: hal/i2c_types.h: No such file or directory
compilation terminated.
src\main.c: In function 'idf_loop':
src\main.c:26:21: warning: unused variable '_err' [-Wunused-variable]
   enum EpdDrawError _err = epd_hl_update_screen(&hl, MODE_GC16, temperature);
                     ^
In file included from .pio\libdeps\esp32dev\epdiy@src-64f018394500b0723813fd6b74fd5859\src\epd_driver\tps65185.c:2:0:
.pio\libdeps\esp32dev\epdiy@src-64f018394500b0723813fd6b74fd5859\src\epd_driver\tps65185.h:4:27: fatal error: hal/i2c_types.h: No such file or directory
compilation terminated.
Compiling .pio\build\esp32dev\FrameworkArduino\Esp.cpp.o
*** [.pio\build\esp32dev\lib1b5\epdiy@src-64f018394500b0723813fd6b74fd5859\epd_driver\pca9555.c.o] Error 1
*** [.pio\build\esp32dev\lib1b5\epdiy@src-64f018394500b0723813fd6b74fd5859\epd_driver\tps65185.c.o] Error 1
```
Probably there's a condition missing somewhere :) Can you try compiling a sketch on a V5 board?

vroland, 2021-09-04:
Yes, looks like I have to fix `#include` paths for the old esp-idf arduino is using. I'll look into it.

vroland, 2021-09-04:
@RealDekkia That is because you are mixing C and C++. The .ino file is C++, and when you link C functions without declaring extern "C", linking will fail. Removing the "extern "C" around idf_setup() and idf_loop() at the top and adding it around the includes `#include "epd_driver.h", #include "epd_highlevel.h"` should help.
Though I should probably extend the headers such that they do that automatically when included from C++.

RealDekkia, 2021-09-04:
Yup, that did it.
Thanks for your help.

mcer12, 2021-09-04:
I have been using it as @vroland suggests but it doesn't work for me in platformio. Platformio seems to use older version of esp-idf so there's probably the issue with #include paths in older esp-idf.

mcer12, 2021-09-04:
Just checked, platformio seems to be using ESP-IDF 4.3 which is the latest. It just can't find the particular `#include "hal/i2c_types.h"`

mcer12, 2021-09-04:
I just commented it out from tps65185.h and pca9555.c and the code compiles fine. Don't know how and why, so feel free to enlighten me :)

vroland, 2021-09-08:
@mcer12 Strange, when I compile with IDF 4.3.1 I don't get such errors...

HJS72, 2021-09-10:
Hi, I have the same problem in platfomio with v6:
`Compiling .pio/build/esp32dev/lib556/epdiy/epd_driver/pca9555.c.o
.pio/libdeps/esp32dev/epdiy/src/epd_driver/pca9555.c:3:27: fatal error: hal/i2c_types.h: No such file or directory`

Can you please check the problem.
THX

martinberlin, 2021-09-10:
Many components of IDF depend on /hal hardware abstraction layer files.
But this is coming when you try to build it with PlatformIO IDE?
Because I do not get this when building the project in command line from Linux with IDF.py 
Can you please give more details on how to reproduce this error?

if you are building this with Platformio and VScode IDE a proper [platformio.ini](https://github.com/martinberlin/cale-idf/blob/master/platformio.ini) should be in place.

mcer12, 2021-09-10:
@martinberlin this happends in Platformio - in arduino framework. Probably has something to do with idf being precompiled there.

martinberlin, 2021-09-11:
Try to search for: i2c_types.h 
In the **.build** directory to understand what is going on. And it happens when compiling any example?

vroland, 2021-09-14:
@mcer12 Could you try again with the current master?

mcer12, 2021-09-14:
@vroland seems to compile fine! What was the culprit?

vroland, 2021-09-14:
This header path seems to be different in arduino. I've made case destinctions in other locations as well. Fortunately, I can just delete this include in this case, because it is automatically included by the i2c driver with the correct path ;)

mcer12, 2021-09-15:
Closing as solved - use the latest epdiy library.

closed 2021-09-15 by mcer12