vroland/epdiy #106: Dragon example won't work repeatedly

mcer12, 2021-09-03:
The sketch should draw dragon again and again, but it only works the first time and then it just blanks white.

```
/* Simple firmware for a ESP32 displaying a static image on an EPaper Screen */

#include "esp_heap_caps.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "epd_driver.h"
#include "epd_highlevel.h"
#include "dragon.h"

EpdiyHighlevelState hl;

void idf_loop() {

  EpdRect dragon_area = {
      .x = 0,
      .y = 0,
      .width = dragon_width,
      .height = dragon_height
  };
  epd_copy_to_framebuffer(dragon_area, dragon_data, epd_hl_get_framebuffer(&hl));

  epd_poweron();
  epd_clear();
  int temperature = 25;
  enum EpdDrawError _err = epd_hl_update_screen(&hl, MODE_GC16, temperature);
  epd_poweroff();

 // while (1) {
      vTaskDelay(5000);
 // }
}

void idf_setup() {
  epd_init(EPD_OPTIONS_DEFAULT);
  hl = epd_hl_init(EPD_BUILTIN_WAVEFORM);
}

#ifndef ARDUINO_ARCH_ESP32
void app_main() {
  idf_setup();

  while (1) {
    idf_loop();
  };
}
#endif

```

martinberlin, 2021-09-04:
If you change epd_clear() for:

    epd_fullclear(&hl, 25);

Does it do the same or draws again?

vroland, 2021-09-04:
Yes, that is the reason :) If you just call epd_clear(), the highlevel driver things the screen is stilll displaying the dragon, thus no change is needed. With epd_fullcelar(&hl, 25), the framebuffer is set to white, and thus the change is registered. I updated the dragon example to be looping.

closed 2021-09-04 by vroland

martinberlin, 2021-09-04:
I think this fact @vroland it’s something to add in the documentation. The highlevel driver updates only when the framebuffer changes

mcer12, 2021-09-04:
Works very well now thanks!