vroland/epdiy #110: Question about gate clock

huming2207, 2021-09-16:
Hi @vroland ,

Firstly, thanks for your great project!

I've got a question after reading your code. I might have missed something, but I would like to know how do you drive the gate clock (the CKV) to make it sync with source data fed in by the 8080 parallel port? 

As far as I understood, all these raw e-paper displays work in these sequences:

1. raise down gate clock
2. send source data (the matched LUT for each pixel) via the parallel port
3. lower down the gate clock, and then next row

I've seen you are using RMT to output the gate signal. But I would like to know how did you manage to keep sync with the parallel signal? Or, is it not necessary to keep the gate and source in sync? If so, then how do we make sure the right source line is written into the right place?

Regards,
Jackson

vroland, 2021-09-19:
Hi,
the CKV signal needs to be active for a specific amount of time (12us) per line. So to output a line, I first program the RMT peripheral to output a pulse of that length, which it does asynchronously. Then, I start the I2S data output. 
But before I output the next line, I check if the RMT is done with that pulse (rmt_busy() in epd_output_row() in display_ops.c). That way, I keep synchronization. 
Does that answer your question?

huming2207, 2021-09-19:
Hi @vroland ,

Yep understood. Thanks.

So then is that also mean, we don't need to keep the gate clock high during source data (from the parallel port) is being transferred? i.e. as long as the gate clock goes up before and during (some portion of) the source data, it should just work?

Regards,
Jackson

vroland, 2021-09-20:
Hi,
Yes, that should be correct for the displays I know. It looks like the source data is first stored into a shift register, which is independent from the output. Then, the latch enable signal transfers that to the output register, which seems to be gated by CKV.

huming2207, 2021-09-21:
@vroland Thanks! Now I got it!

closed 2021-09-21 by huming2207