vroland/epdiy #111: How to update a small part of the display when coming out of deep sleep

cgreening, 2021-09-20:
More of a question, than an issue.

I have a UI where I'm drawing a box around an item to indicate it is selected.

When I come out of deep sleep I just want to move the position of the box without redrawing everything on the screen.

This works well without going into deep sleep as I can erase the previous box by drawing over it in white and then draw the new box.

When coming out of deep sleep it doesn't work - the new box is drawn, but the old box remains - I'm assuming this is because the display driver doesn't think the screen area has changed so doesn't update it.

Is there a way to invalidate part of the screen so that it will get flushed to the display even if it doesn't seem to have any changes?

vroland, 2021-09-20:
Hi,
this is indeed a bit tricky. With the highlevel API, the driver compares the front buffer to a back buffer to see if something has changed. But the back buffer is of course not preserved during deep sleep. 
I think the easiest option would be to write the old display state into the back buffer (see https://github.com/vroland/epdiy/blob/master/src/epd_driver/include/epd_highlevel.h#L63) before writing the new state and then update as normal. 
Otherwise, you could use the low-level API to manually erase the old box, but mixing the two can lead to confusions.
Hope that helps :)

cgreening, 2021-09-20:
Makes sense.

I've got an SDCard connected so I could easily save out the current front and back buffer before sleeping. I'll do some tests and see if that is faster than simply redrawing everything.

closed 2021-09-20 by cgreening