vroland/epdiy #112: epd_temperature_init conflicts with LilyGoT54.7 and GPIO_NUM_35

cgreening, 2021-09-20:
I'm using a LilyGoT54.7 board and want to use the buttons - they are connected to `GPIO_NUM_34`, `GPIO_NUM_39` and `GPIO_NUM_35`.

These buttons work fine unless I call `epd_init` after calling that the behaviour becomes quite weird.

`GPIO_NUM_35` is always low when I read from it.

I think this is because the code is trying to use GPIO_NUM_35 to read temperature. Possibly there's a missing preprocessor directive to not do this when either of the these are defined:

```
-DCONFIG_EPD_DISPLAY_TYPE_ED047TC1
-DCONFIG_EPD_BOARD_REVISION_LILYGO_T5_47
```

I'm not sure what should be returned from these functions if there is not temperature sensor and if it will break anything if they are replaced with dummy functions for the LilyGo board.

hpsaturn, 2021-09-25:
Hi, I have maybe a similar situation. With the library from the LilyGo official repository of EPD47 I have right readings of battery level with USB or only with the battery. But with this library, the original library that LilyGO have a fork, I can't read the battery level (ADC) on pin 36 and also pin 35. It only works with USB. Maybe do you know is it a same issue?

cgreening, 2021-09-25:
Make sure you have called both:

```
epd_init(EPD_OPTIONS_DEFAULT);
```

And

```
epd_poweron();
```

It should work if you do both those calls and then try and measure the battery.

hpsaturn, 2021-09-25:
Thanks for your comment. But I already have this flow. I think that it is a different problem, because the official demo fails too like my behavior. The demo only works with USB, with battery not.. I'm going to publish the issue.

cgreening, 2021-09-25:
This seems to work - https://github.com/atomic14/lilygo-epd47-battery

Should probably move off this issue - feel free to open on that repo if it doesn't work.

hpsaturn, 2021-09-25:
It is true, I wrong, because I had a CPU frequency with other value, for this reason in battery mode sometimes the ADC only reached ~2v to 2.9v

Here, in a [Telegram group](https://t.me/esp_es/14163) of ESP32, I showed the last version of my work, works nice!

Thanks!
[image]

PR #113 (2021-10-04): Prevent conflict with buttons on LilyGOT54.7 boards
Fixes: #112 

On the LilyGoT5 4.7 boards GPIO35 is connected to a button with a pullup resistor so can't be used for reading temperature values.

I've added a `#define` to provide dummy functions.
closed 2021-10-04 by vroland