vroland/epdiy #116: Looking for people to test an ePub reader

cgreening, 2021-09-29:
Hope it's ok to post this here - I've been coding up an ePUB reader and it's got to the point where it "good enough".

I've only got a LilyGO 4.7 board and was wondering if anyone could try it on their boards to see if it works...

https://github.com/atomic14/diy-esp32-epub-reader

martinberlin, 2021-09-30:
I will give it a try!
I think you will be better off using the https://epdiy.slack.com channels since there many people that use this component hang out and share stuff.

cgreening, 2021-09-30:
Slack button doesn't seem to work anymore - says the link is no longer valid :( - can someone invite me?

martinberlin, 2021-09-30:
Sure Chris, just send me your email: martin@luckycloud.de

cgreening, 2021-09-30:
Moved to slack

closed 2021-09-30 by cgreening