vroland/epdiy #121: epd_deinit() not work properly

stig3, 2021-10-21:
I was testing low-power of epdiy
on my board when I use esp_deep_sleep_start() to sleep ESP32 directly, the current of deep sleep is about 5.5uA.
but after I display any image then call epd_deinit() and esp_deep_sleep_start(), the sleep current raise to 50uA.
So is there any way to fully deinit epd?

mcer12, 2021-11-27:
@stig3 aren't you using ESP-WROVER-E by any chance?

stig3, 2021-11-29:
@mcer12 Thanks. I was using ESP-WROVER-I(8Mb).

mcer12, 2021-11-29:
@stig3 If you're using different module than WROVER-B you can try rtc_gpio_isolate(GPIO_NUM_12); right before going to sleep. I don't remember the caviats of V4 design but there could be more issues. For real low power (<15uA), go for V5 version.

mcer12, 2021-11-29:
@vroland I know there's a little note about using B module on wiki but probably should be more visible in the as it's very important for low power. :)

martinberlin, 2022-05-14:
I think about this, a part of the WROVER version, there has been some new code merged that disconnects the IOs when epdiy is deinitialized both for V4 and Lilygoboards.

martinberlin, 2022-05-15:
@stig3 can you please pull latest master and try it to see if something changed regarding low power consumption?

martinberlin, 2022-06-21:
Closing here because there was lack of feedback since at least 2 months from the last interaction.
IMHO if there is 2 entire months without any reply, we can consider the issue as stale, and close it. There is no benefit in keeping Issues open eternally if the author of the issue does not interact with it.

closed 2022-06-21 by martinberlin