vroland/epdiy #123: LilyGo-T5 4.7 displayed faded in patches!

chrisb2, 2021-11-07:
The text I am writing to my display is faded quite badly in patches, particularly on the left hand side (landscape). Has anyone got any solution to this?

My code (Arduino for ESP32) is a follows. I am not using deep sleep (just delay()). Once the delay expires the display code is run again
**Setup**
```
  epd_init(EPD_OPTIONS_DEFAULT);
  m_display = epd_hl_init(EPD_BUILTIN_WAVEFORM);
  epd_set_rotation(EPD_ROT_LANDSCAPE);
  m_framebuffer = epd_hl_get_framebuffer(&m_display);
  epd_clear();
```
**Display**
```
  epd_hl_set_all_white(&m_display);
  # code which write text etc to the display is here
  epd_poweron();
  enum EpdDrawError err = epd_hl_update_screen(&m_display, MODE_GC16, 20);
  delay(500);
  epd_poweroff();
```

vroland, 2021-11-11:
Is there a reason why you use delay() before poweroff? Also, does this happen with the component provided by LILYGO as well?

chrisb2, 2021-11-17:
I just tried delay(), but same thing was happening with deep sleep. I will check with one of the existing components, I did examine these and try to make my code consistent, but maybe I missed something, so good to test one of these.

chrisb2, 2021-12-21:
Checkout latest code and made some small change to be consistent with example, now working well.

Thanks

closed 2021-12-21 by chrisb2