vroland/epdiy #125: About show jpeg

lixy123, 2021-11-13:
The demo of "/Example/www-image" is very good. 
Can you transplant the code to the ink screen API separately?

martinberlin, 2021-11-13:
Sorry but I don’t really get what is the task here.

lixy123, 2021-11-14:
yes, add some api of do with jpg image format , 
for example,  showjpg_byfile(), showjpg_byurl() to core api of EPDiy

martinberlin, 2021-11-14:
Sorry, I did the example with the ideas that @vroland gave me but I'm not going to go that far. As you can see it involves quite a lot of different parts doing this using esp-idf like the request itself but separately the callback function to read incoming data. And additionally the SSL certificate if your urls is an http secure request.  That is complex to add and I'm also not so sure if it's intended to be on the core that is supposed to be an eink controller, with geometry functions and don't support but so far image decompression is not in the mix.
But if you have the idea to do it please go ahead and make a pull request, we will analyze it and discuss if it could be added like this.

vroland, 2021-11-14:
I see it in a similar way. The whole networking thing and JPEG decompression is rather complex and resource hungry, and doesn't really have anything to do with driving e-paper displays. I would suggest to move this into a separate library, that allows to hook into a pixel drawing function, so it can be used with epdiy.

martinberlin, 2021-11-14:
Thanks for your feedback. @lixy123 maybe what we can do is to close this issue and plan to make an epdiy-image component that can add this functionality.
It sounds like an interesting addition that if it’s done well can help a some people to make image handling easier. You can find my email in my GitHub profile.

lixy123, 2021-11-15:
Thank you for your feedback
I suggest adding only JPG read in and displayed from spiff.
If do it , When displaying some small icon images, you don't have to deal with the image binary stream manually

lixy123, 2021-11-15:
[LilyGo-EPD47-ShowJPG.zip](https://github.com/vroland/epdiy/files/7537083/LilyGo-EPD47-ShowJPG.zip)

I copy you code of show jpg image example, write one library code,  
If you find it useful, consider incorporating it into the core API, 
Code not tested.
good luck!
code source : https://github.com/vroland/epdiy/blob/master/examples/www-image/main/jpg-render.c

lixy123, 2021-11-15:
haha,
I debugged the code at night and it can run normally.
Put into GitHub already: https://github.com/lixy123/LilyGo-EPD47-ShowJPG

martinberlin, 2021-11-16:
Thanks for the example @lixy123 

Soon I will start a component to parse JPG/PNG following the idea of @vroland .
So it will accept a callback function like **epd_draw_pixel** and then it will send the x, y and color parameters when parsing the image.

Please close this ticket since the relevant parts have been discussed and at this point has nothing to do with EPDiy anymore.

closed 2021-11-17 by lixy123