vroland/epdiy #126: V6 in release lack of R7 and D7

waitlamp, 2021-11-23:
Hello, 
Thank you for this interesting project. I have orderd some SMT boards for JLC before carefully check (with Coupons😂).
They asked me why my SMT position file have a R7 resistor without pad.
I relized I used master branch to generate BOM and pos but Gerber file is the V6 in release.
in 1647406e626fd9eba89177740b7ece9a50d661cc R7 and D7 was added and V6 file has updated, but in Release it was not.

now they have manufactured. Does this difference pose any problem? 

Thanks!

vroland, 2021-11-23:
Arg, sorry, I should have updated this. You may need to add a bodge diode for running from battery to work propery:
[image]

vroland, 2021-11-23:
I've added a new release with the updated board. But at least with the bodge diode shown in the picture, there should be no further issues.

closed 2021-11-23 by waitlamp