vroland/epdiy #128: failed: NoMemory

M737M737, 2021-11-28:
Hi,
I ran into an error "deserializeJson() failed: NoMemory"
Do you know how to fix it?

-------------------
10:20:40.373 -> WiFi connected at: 10.0.0.125
10:20:40.408 -> ***Current time***
10:20:41.823 -> Sun Nov 28 2021   11:20:39
10:20:41.863 -> 
10:20:41.898 -> Creating object...and deserializeJson() failed: NoMemory
10:20:41.953 -> 
10:20:41.974 -> Creating object...and deserializeJson() failed: NoMemory
10:20:42.028 -> 
10:20:42.047 -> Creating object...and deserializeJson() failed: NoMemory
10:20:42.101 -> 
10:20:42.121 -> Creating object...and deserializeJson() failed: NoMemory
10:20:42.174 -> I (8329) wifi:state: run -> init (0)
10:20:42.233 -> I (8330) wifi:pm stop, total sleep time: 3258271 us / 4599927 us
10:20:42.332 -> 
10:20:42.332 -> W (8331) wifi:<ba-del>idx
10:20:42.332 -> I (8334) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
10:20:42.461 -> I (8341) wifi:flush txq
10:20:42.461 -> I (8342) wifi:stop sw txq
10:20:42.461 -> I (8345) wifi:lmac stop hw txq
10:20:42.511 -> I (8347) wifi:Deinit lldesc rx mblock:10
10:20:42.602 -> EPD clear took 1078ms.
10:20:43.229 -> Icon name: 
10:20:43.271 -> Icon name: 
10:20:43.308 -> Icon name: 
10:20:43.338 -> Icon name: 
10:20:43.369 -> Icon name: 
10:20:43.408 -> Icon name: 
10:20:43.448 -> Icon name: 
10:20:43.486 -> Icon name: 
10:20:43.521 -> Entering 3600-secs of sleep time
10:20:44.936 -> Awake for : 9.648-secs
10:20:44.936 -> Starting deep-sleep period...

martinberlin, 2021-11-28:
Never run into this. Questions:
Can you update your entry with the Serial from the beginning to check that PSRAM is being initialized property?
Also if possible upload the full code somewhere in github and post a link here to it so we can take a look. Otherwise it's quite hard to understand what is going on.

M737M737, 2021-11-28:
----
10:20:35.742 -> [0;32mI (1228) cpu_start: Pro cpu start user code[0m
10:20:35.749 -> [0;32mI (1228) cpu_start: cpu freq: 240000000[0m
10:20:35.782 -> [0;32mI (1228) cpu_start: Application information:[0m
10:20:35.793 -> [0;32mI (1231) cpu_start: Project name:     weather[0m
10:20:35.799 -> [0;32mI (1236) cpu_start: App version:      v6-dirty[0m
10:20:35.813 -> [0;32mI (1241) cpu_start: Compile time:     Aug 27 2021 17:15:51[0m
10:20:35.835 -> [0;32mI (1247) cpu_start: ELF file SHA256:  3359a2f7c5b8d7bd...[0m
10:20:35.862 -> [0;32mI (1253) cpu_start: ESP-IDF:          v4.4-dev-2740-gf65c8249af[0m
10:20:35.880 -> [0;32mI (1260) heap_init: Initializing. RAM available for dynamic allocation:[0m
10:20:35.910 -> [0;32mI (1267) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
10:20:36.016 -> [0;32mI (1273) heap_init: At 3FFBB820 len 000247E0 (145 KiB): DRAM[0m
10:20:36.061 -> [0;32mI (1279) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
10:20:36.099 -> [0;32mI (1286) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
10:20:36.123 -> [0;32mI (1292) heap_init: At 4009C070 len 00003F90 (15 KiB): IRAM[0m
10:20:36.145 -> [0;32mI (1299) spiram: Adding pool of 4095K of external SPI memory to heap allocator[0m
10:20:36.178 -> [0;32mI (1307) spi_flash: detected chip: generic[0m
10:20:36.200 -> [0;32mI (1311) spi_flash: flash io: qio[0m
10:20:36.212 -> [0;32mI (1317) cpu_start: Starting scheduler on PRO CPU.[0m
10:20:36.238 -> [0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
10:20:36.253 -> [0;33mW (1331) main: Hello World!
10:20:36.267 -> [0m
10:20:36.267 -> Heap summary for capabilities 0x00000800:
10:20:36.295 ->   At 0x3ffae6e0 len 6432 free 1132 allocated 3148 min_free 128
10:20:36.323 ->     largest_free_block 512 alloc_blocks 32 free_blocks 1 total_blocks 33
10:20:36.352 ->   At 0x3ffbb820 len 149472 free 131328 allocated 15992 min_free 131328
10:20:36.382 ->     largest_free_block 131072 alloc_blocks 31 free_blocks 1 total_blocks 32
10:20:36.427 ->   At 0x3ffe0440 len 15072 free 12920 allocated 0 min_free 12920
10:20:36.456 ->     largest_free_block 8192 alloc_blocks 0 free_blocks 1 total_blocks 1
10:20:36.489 ->   At 0x3ffe4350 len 113840 free 111688 allocated 0 min_free 111688
10:20:36.519 ->     largest_free_block 65536 alloc_blocks 0 free_blocks 1 total_blocks 1
10:20:36.551 ->   At 0x4009c070 len 16272 free 14120 allocated 0 min_free 14120
10:20:36.581 ->     largest_free_block 8192 alloc_blocks 0 free_blocks 1 total_blocks 1
10:20:36.629 ->   Totals:
10:20:36.629 ->     free 271188 allocated 19140 min_free 270184 largest_free_block 131072
10:20:36.659 -> Heap summary for capabilities 0x00000400:
10:20:36.689 ->   At 0x3f800000 len 4194303 free 4192151 allocated 0 min_free 4192151
10:20:36.720 ->     largest_free_block 2097152 alloc_blocks 0 free_blocks 1 total_blocks 1
10:20:36.752 ->   Totals:
10:20:36.768 ->     free 4192151 allocated 0 min_free 4192151 largest_free_block 2097152
10:20:36.810 -> [0;32mI (1437) esp_temperature: Characterized using eFuse Vref
10:20:36.848 -> [0m
10:20:36.848 -> [0;33mW (1440) main: allocating...
10:20:36.865 -> [0m
10:20:36.865 -> 
10:20:36.909 -> Connecting to: Ziege
10:20:36.909 -> I (1467) wifi:wifi driver task: 3ffd4950, prio:23, stack:6656, core=0
10:20:36.966 -> I (1492) wifi:wifi firmware version: 70f0192
10:20:37.001 -> I (1492) wifi:wifi certification version: v7.0
10:20:37.022 -> I (1493) wifi:config NVS flash: enabled
10:20:37.063 -> I (1493) wifi:config nano formating: disabled
10:20:37.082 -> current temperature: 20.500000
10:20:37.099 -> I (1497) wifi:Init data frame dynamic rx buffer num: 32
10:20:37.133 -> I (1505) wifi:Init management frame dynamic rx buffer num: 32
10:20:37.167 -> I (1510) wifi:Init management short buffer num: 32
10:20:37.201 -> I (1515) wifi:Init static tx buffer num: 16
10:20:37.220 -> I (1519) wifi:Init tx cache buffer num: 32
10:20:37.243 -> I (1523) wifi:Init static rx buffer size: 1600
10:20:37.292 -> I (1527) wifi:Init static rx buffer num: 10
10:20:37.313 -> I (1531) wifi:Init dynamic rx buffer num: 32
10:20:37.354 -> [0;32mI (1535) wifi_init: rx ba win: 6[0m
10:20:37.373 -> [0;32mI (1538) wifi_init: tcpip mbox: 32[0m
10:20:37.413 -> [0;32mI (1543) wifi_init: udp mbox: 6[0m
10:20:37.433 -> [0;32mI (1546) wifi_init: tcp mbox: 6[0m
10:20:37.451 -> [0;32mI (1550) wifi_init: tcp tx win: 5744[0m
10:20:37.471 -> [0;32mI (1554) wifi_init: tcp rx win: 5744[0m
10:20:37.471 -> [0;32mI (1559) wifi_init: tcp mss: 1440[0m
10:20:37.471 -> [0;32mI (1563) wifi_init: WiFi IRAM OP enabled[0m
10:20:37.496 -> [0;32mI (1567) wifi_init: WiFi RX IRAM OP enabled[0m
10:20:37.543 -> [0;32mI (1573) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
[... 26 lines cut ...]
10:20:42.174 -> I (8329) wifi:state: run -> init (0)
10:20:42.233 -> I (8330) wifi:pm stop, total sleep time: 3258271 us / 4599927 us
10:20:42.332 -> 
10:20:42.332 -> W (8331) wifi:<ba-del>idx
10:20:42.332 -> I (8334) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
10:20:42.461 -> I (8341) wifi:flush txq
10:20:42.461 -> I (8342) wifi:stop sw txq
10:20:42.461 -> I (8345) wifi:lmac stop hw txq
10:20:42.511 -> I (8347) wifi:Deinit lldesc rx mblock:10
10:20:42.602 -> EPD clear took 1078ms.
10:20:43.229 -> Icon name: 
10:20:43.271 -> Icon name: 
10:20:43.308 -> Icon name: 
10:20:43.338 -> Icon name: 
10:20:43.369 -> Icon name: 
10:20:43.408 -> Icon name: 
10:20:43.448 -> Icon name: 
10:20:43.486 -> Icon name: 
10:20:43.521 -> Entering 3600-secs of sleep time
10:20:44.936 -> Awake for : 9.648-secs
10:20:44.936 -> Starting deep-sleep period...
10:20:44.978 -> 
------------

It's the original weather example code (no changes)

martinberlin, 2021-11-28:
Looks strange. maybe the arduinoJson version has something to do?
Will try to compile it again on Monday and let you know what I've got. But for me it was working some months ago.

M737M737, 2021-11-28:
Thank you!

Yes, the API key should be valid, I've checked it again today.

martinberlin, 2021-11-29:
I tried to compile it and I succeeded but I'm getting a different error so I could not test it:

Connecting to: WLAN-724300
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4008de68
0x4008de68: _esp_error_check_failed at /home/martin/esp/esp-idf/components/esp_common/src/esp_err.c:45

file: "/home/martin/esp/projects/epdiy-fork/src/epd_driver/tps65185.c" line 21
**This line is: ESP_ERROR_CHECK(i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS));**
func: i2c_master_read_slave
expression: i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS)

abort() was called at PC 0x4008de6b on core 1

@vroland I'm using IDF 4.2 do you have any idea why this could be happening?
Using 4.3 I didn't hit this so far.

vroland, 2021-11-30:
Hm, could be the case. Or maybe it's some menuconfig setting? If you copy over the sdkconfig, does it change anything?

M737M737, 2021-11-30:
Thanks for the possible error causes. I'll try to renew my Arduino Component and Arduino JSON.
Is there a new download link?
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
The one of the readme file doesn't work anymore...

I have to admit that I'm not experienced with ESP-IDF, so maybe I did a mistake at building.

M737M737, 2021-11-30:
@vroland my changes in menuconfig: Display Type, board revision, partition (to partitions.csv)

Is there anything else to change?

---
Right now I'm running into a build error:

[image]

Is this error known?

vroland, 2021-11-30:
Hm, I didn't encounter this error yet. But building the arduino component tricky... 
It is probably some incompatibility between the versions.
Maybe we should change the weather example into an arduino IDE example, that could make it easier. But someone needs to do it :/

martinberlin, 2021-12-01:
Hi Valentin,
tried that copying dragon/sdkconfig to weather (And adding the correct partitions.csv with menuconfig) but it does exactly the same. There is a conflict when trying to instantiate I2C.
This is only with V6, with V5 and with Lilygo seems to work correctly and I now my father runs the weather in his V5 board. 
@M737M737 I will compile tomorrow the example in the Lilygo and see if I can reproduce your error.

M737M737, 2021-12-01:
Hi,

I installed ESP-IDF 4.4 and the newest Arduino Core ESP32 (https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html).
My changes in menuconfig: Display Type, board revision, partition (to partitions.csv)
In my case: ED097OC4, Board V4

With this configuration I managed to compile the weather example sucessfully.

It seems that ArduinoJson is not necessary anymore.

martinberlin, 2021-12-01:
Great! Then this has to be written somewhere in the readme, I think this example needs to be co piled with IDF >= 4.3 and yes I don’t understand why but arduinoJson is not anymore needed.
I would love to have this example IDF only without any further arduino dependencies, but as is copyrighted I don’t know if would be ok to do that modifying the original.

closed 2022-02-27 by M737M737