vroland/epdiy #130: ED097OC4 doesn't work in default drawing mode

mcer12, 2021-12-03:
I know this was discussed in a different issue and I'm not the only one having this issue, but I fail to find the thread... maybe it was only mentioned on slack.

ED097OC4 doesn't work properly in default mode, the picture is washed out and has extremely low contrast with bleeding edges. This happends with both MODE_GC16 and MODE_GL16. There's probably some issue with timing because it seems like it's going correctly but at the end it "overshoots" and goes to almost white. I have to ramp up VCOM to over -2V (display sticker says -1.31V) to see anything.

I managed to resolve it using MODE_EPDIY_WHITE_TO_GL16, with this the picture appears perfect with the correct VCOM. There is very barely noticable white bleed around the edge.

mcer12, 2021-12-05:
This issue also seems to happen on TC2 with vendor waveform. Right at the end of the update, the image fades a bit.

vroland, 2021-12-30:
I'll try out the vendor waveforms more some time and see if it's a bug or just epdiy being not suitable for these waveforms.

mcer12, 2021-12-30:
There's definitely a bug (at least with the 4bpp waveforms) as the script doesn't output working header file. I was looking at the code but this is way out of my scope :D

vroland, 2021-12-30:
Ok, good to know. Though I'm afraid that that's not my top priority right now. There's some information in the project logs of this project: https://hackaday.io/project/21168-fpga-eink-controller/

martinberlin, 2022-06-21:
I have a 9.7" display but I'm not sure if it's ED097OC4 since they usually do not come with the model number printed. But I will make some tests and try to find out what is going on.

martinberlin, 2022-07-07:
Hi @mcer12 wanted to get one of this ED097OC4(LF) displays for my father’s birthday
https://m.es.aliexpress.com/item/1005003493801117.html Would this work to reproduce this?

martinberlin, 2024-04-25:
Hello Martin @mcer12 I'm buying this month one of this OC4 9.7" displays. 
Can you tell me how to reproduce it?  Just any image even the Dragon example will fade at the end?

mcer12, 2024-04-25:
@martinberlin holy moly, it's been two years :D I'll see if I can replicate it and will try to make a 60fps video of it, how it behaves.
"Just any image even the Dragon example will fade at the end?" - exactly yes. The display does its thing, flashes, transitions seemingly correctly and at the end it "overshoots" and appears washed out.

mcer12, 2024-04-25:
@martinberlin also check slack, there's a relevant discussion about these displays. It's very difficult to get the exact display model you need.