vroland/epdiy #133: Arduino IDE compile/upload errors caused `boards.local.txt`

Sam-Izdat, 2021-12-16:
Arduino IDE 1.8.18
Board: LILYGO-T5 4.7 inch
OS: Windows

Followed the documentation to set it up epdiy as Arduino IDE library (side note: apparently boards.txt directory is now in `C:\Users\[User]\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1` which took a while to locate). Copied boards.local.txt. The boards list populated, but compiling any sketch raised the error:

    exec: "--elf-gcc.path}/bin/--elf-g++": file does not exist
    Error compiling for board LILYGO T5-4.7 inch e-paper.

Adding the following missing lines to boards.local.txt before `[board].build.mcu` line resolved the compilation error:

   lilygo_t5_47.build.tarch=xtensa
   lilygo_t5_47.build.bootloader_addr=0x1000
   lilygo_t5_47.build.target=esp32

An upload error followed, complaining about upload flags. Adding the following missing lines to boards.local.txt after `[board].upload.wait_for_upload_port` line resolved the upload error:

    lilygo_t5_47.upload.flags=
    lilygo_t5_47.upload.extra_flags=

I assume these errors need to be resolved for every board, but I have not tested this.

vroland, 2021-12-30:
Good to know! It might also have something todo with the new 2.0.0 version of the arduino esp32 core. Would you mind submitting a pull request with the required changes?

Sam-Izdat, 2021-12-30:
Not at all! I'll try and get it in this weekend. I'm unable to to test the other boards fully, so will have to just assume the uploaf fix is universal unless someone can check and finds otherwise.

Sam-Izdat, 2022-01-05:
Put in pull request #135 - using master branch. I can resubmit if you need this on another branch instead.

vroland, 2022-01-06:
Thanks, merged :)

closed 2022-01-06 by vroland