vroland/epdiy #134: Display not responding with ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400887cc

Waged, 2021-12-28:
- Found this error after succeeding to program the V6 board & enabling SPI-RAM & changing the VCOM value to the one written to the screen.

`At 0x400907d8 len 63528 free 61376 allocated 0 min_free 61376
    largest_free_block 32768 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 341588 allocated 18804 min_free 340584 largest_free_block 131072
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194303 free 4192151 allocated 0 min_free 4192151
    largest_free_block 2097152 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 4192151 allocated 0 min_free 4192151 largest_free_block 2097152
Dimensions after rotation, width: 1024 height: 758
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400887cc
file: "C:/Users/X299/Downloads/epdiy-master/epdiy-master/src/epd_driver/config_reg_v6.h" line 105
func: cfg_poweron
expression: tps_write_register(reg->port, TPS_REG_ENABLE, 0x3F)

abort() was called at PC 0x400887cf on core 0

Backtrace:0x4008a439:0x3ffb8db0 0x4008ac89:0x3ffb8dd0 0x4008f759:0x3ffb8df0 0x400887cf:0x3ffb8e60 0x400d73b1:0x3ffb8e80 0x400d6235:0x3ffb8ea0 0x400d651a:0x3ffb8f30 0x400eb0a2:0x3ffb8f50

ELF file SHA256: 7757b14bf033fd56

Rebooting...
ets Jun  8 2016 00:22:57
`

martinberlin, 2021-12-29:
Just a hint since I'm not an expert in this topic, it seems to fail when trying to communicate with the  PMIC for E-Ink tps65185. Is not throwing any other error with I2C instantiation before this error you show?
What value of VCOM you are using?
Leave the default just to see if it's not the value the problem.

vroland, 2021-12-30:
Yes, that looks like the I2C communication with the TPS65185 failed. Are you sure the I2C pins are soldered correctly?
Since the power good line seems to have come up, it does not look like the chip failed completely or is in power save mode.

Waged, 2022-01-02:
it could be not soldered correctly but I soldered other one and seems to be other problem as the TPS65185  got heated and nothing happen to the screen.
is there a way to check if it's well soldered ? test points that i can know from or even a test board that i can directly order to avoid soldering issues.. 
- one key note to tell that I'm using WROVER not WROVER B  with 8 MB of PSRAM and enabled and everything looks nice.
my screen is the ED060XC3 .

vroland, 2022-01-06:
Hm, that is weird, since aparently the chip outputs power good before that happens. 
Can you e.g. add a delay before the failing statements and measure the test points?
But also, the TPS65185 should not really heat up in normal operation...

closed 2022-05-10 by martinberlin