vroland/epdiy #137: Is there a recommended ui library to use with epdiy?

lanistor, 2022-01-13:
Such as lvgl?
Not like [martinberlin/lvgl_epaper_drivers](https://github.com/martinberlin/lvgl_epaper_drivers), this library rendering so slow, and cannot resolve font' burrs, may be it's because it using `epd_hl_update_area` to render for anything, and need to convert every pixel before render.

epd_driver's API is much like OpenGL, so it's hard to develop a big project without ui library.

martinberlin, 2022-01-13:
Hello Lanistor, you need to use epd_hl_update_area to render because that is how LVGL works!
It updates small parts only for the UI elements that change. But if you play around with colors and use only black and different updates modes it works reasonably well (At least for me trying with LILYGO EPD47)
Of course it's meant for faster displays...
There are more comments here: https://github.com/martinberlin/lv_port_esp32-epaper :

> 9. For [EPDiy](https://github.com/vroland/epdiy) board, Make sure to select the correct board revision and display type in `Components E-Paper Driver`.
> 
>     9.1. Select  `(X) <LV_THEME_DEFAULT_INIT_MONO> Default init for mono theme` in ` Component config → LVGL configuration → Theme usage → Select theme default init`.   
> 
>     9.2. Uncheck all in `Component config → LVGL configuration → Theme usage → Enable theme usage`. 

Also I stopped updating my LVGL port since I realized it will never be merged due to a License conflict (LVGL uses MIT License) while this project GPL v3, that are not compatible.

lanistor, 2022-01-13:
> It updates small parts only for the UI elements that change. But if you play around with colors and use only black and different updates modes it works reasonably well (At least for me trying with LILYGO EPD47)

Thanks a lot.
I have edit my configuration as docs you supplied, and using v5 board of epdiy, but it still cost more than 10 seconds to render a block that has black background color, here's the demo (the rect block of left side ):

[image]

martinberlin, 2022-01-13:
Sorry but I never tried with such a big display. Maybe there is a way to enlarge more the internal buffer so it updates bigger parts each time, but you need to understand well LVGL to try this kind of stuff (Took me a while and still know 20% of it)
Also in my EPD47 also took like 3-4 seconds to update even using fast MODE_DU update, so it might very well be that takes this time. Anyways comment Serial debugging and optimize a bit whenever possible. 
Also check in menuonfig that ESP32 is at full 240 Mhz full speed:
**→ Component config → ESP32-specific → CPU Frequency**

It can be optimized a bit more for sure. But there is no magic, if it takes 1 second to make a full update, it will take longer to the same update in 20 different partial updates. This LVGL port was though to be used in small UX and was highly experimental. I really think at this point it will be very difficult to have a full UI framework working in a usable speed with EPDiy. It's simply too much updates / interruptions to be handled. 
@vroland @mcer12  if you have any other optimization tips please comment. Thanks in advance!

lanistor, 2022-01-17:
> Sorry but I never tried with such a big display. Maybe there is a way to enlarge more the internal buffer so it updates bigger parts each time, but you need to understand well LVGL to try this kind of stuff (Took me a while and still know 20% of it) Also in my EPD47 also took like 3-4 seconds to update even using fast MODE_DU update, so it might very well be that takes this time. Anyways comment Serial debugging and optimize a bit whenever possible. Also check in **idf.py menuconfig** that ESP32 is at full 240 Mhz full speed: **→ Component config → ESP32-specific → CPU Frequency**
> 
> It can be optimized a bit more for sure. But there is no magic, if it takes 1 second to make a full update, it will take longer to the same update in 20 different partial updates. This LVGL port was though to be used in small UX and was highly experimental. I really think at this point it will be very difficult to have a full UI framework working in a usable speed with EPDiy. It's simply too much updates / interruptions to be handled. There is an additional thing that I need to try: if the partial buffer is all white, it should simply skip it ( need to try this )
> 
> Note that I have also an epaper phone Hisense A7 CC that has a much faster multi CPU processor and quite sure a 16 data lines EPD connected in parallel and UX is working ok but still not so fast as a TFT phone and it tooḱ me weeks to get accustomed to it. Make sure also to read the [Drawing section in LVGL documentation](https://docs.lvgl.io/latest/en/html/overview/drawing.html). The defined buffers size can be changed if you have enough free RAM to do it. That might speed up things but again, is not magic! @vroland @mcer12 if you have any other optimization tips please comment. Thanks in advance!

I found a way to resolve it: edit lvgl's library, and change drawing functions directly to epdiys', such as: lv_draw_rect and so on. 
I had tested it  for lv_draw_rect, it works, and rendering fast enough. 

But the font drawing function is a little hard to match.

martinberlin, 2022-01-17:
Ok if you have success with that please publish your changes in a fork since I would also like to give it a try.

lanistor, 2022-01-17:
OK, i will.

martinberlin, 2022-02-26:
@lanistor is there any other pending thing in this issue or we can close here?
I would be still interested in trying out your fork to see how fast it goes.

lanistor, 2022-02-27:
@martinberlin We are still using `epd_hl_update_area` for rendering, for using origin epdiy's drawing functions is really hard to match with lvgl, especially for font functions. And we increased the buffer of lvgl to the screen size, it can run faster, now a full screen rendering costs 3-4 seconds. We will continue to optimize it.

closed 2022-02-27 by lanistor