vroland/epdiy #138: Why use GPL v3 license? not MIT?

lanistor, 2022-01-13:


vroland, 2022-01-14:
The main reason is that I wanted derivative works to stay open-source. This, for example, has helped with maintaining compatibility with the LILYGO board, that is based on epdiy. I hindsight, I probably should have chosen LGPL instead, but not having much embedded experience I didn't think of that.

lanistor, 2022-01-15:
@vroland Thanks for your explain.
Will you change it to LGPL?

We are planing to use it in our hardware product, but we could not open source of our project code.
It's really a wonderful library, we really what to use it.

vroland, 2022-01-17:
Since by now a number of people have contributed, I'd have to ask them all if they are Ok with re-licensing to LGPL, which is a bit cumbersome. According to Github, there are 18 contributors right now. If you are really keen and get everyone to accept (e.g. write a post here) I would be fine with changing to LGPL.

martinberlin, 2022-01-17:
I would have no objection with a License update.

>Many of the most common free-software licenses, especially the permissive licenses, such as the original MIT/X license, BSD licenses (in the three-clause and two-clause forms, though not the original four-clause form), MPL 2.0, and LGPL, are GPL-compatible.

This will also enable my contribution to use EPDiy with LVGL possible since it is not going to be merged if this project stays with GPL v3. So now there are 17 contributors to ask. For some of them it would be maybe faster to ask in the https://epdiy.slack.com chat group.

lanistor, 2022-01-17:
Or directly refer(@) them in this issue, or a new issue? Then waiting for their reply.

martinberlin, 2022-01-18:
>Since by now a number of people have contributed, I'd have to ask them all if they are Ok with re-licensing to LGPL, which is a bit cumbersome.

Other contributors in order of appearance in https://github.com/vroland/epdiy/graphs/contributors
@sebastius
@mmib
@jdoubleu
@markbirss
@mcer12
@cgreening
@fdev31
@SimonSelg 
@BackSlasher
@Li-Tianming 
@holgerlembke
@entozoon 

Please give your opinion of this should be done or stays how it is. I'm fine with either decision since we need to respect all opinions and collaborations in an open source project.
Valentin if you find some time can you please give me some feedback about #131

BackSlasher, 2022-01-18:
I don't mind.
Thanks for asking and bigger thanks for maintaining this project

holgerlembke, 2022-01-18:
Don't mind, too (with my tiny bitsy contribution....).

cgreening, 2022-01-18:
I've got no problem with the license change. LGPL makes it much easier for people to use.

sebastius, 2022-01-18:
I’m okay with this!

Sebastius

entozoon, 2022-01-18:
Definitely, the more open the better :)

fdev31, 2022-01-18:
Same for me, +1 :)

jdoubleu, 2022-01-18:
Thanks for asking!
I'm totally fine with `LGPL`.

mcer12, 2022-01-18:
No problem at all with me.

markbirss, 2022-01-19:
I'm also OK with it going more open

Li-Tianming, 2022-01-19:
It's OK for me 🤪

vroland, 2022-01-19:
Thanks, this is looking good so far!

I think @DavidM42 and @Sam-Izdat are missing.

DavidM42, 2022-01-19:
Im okay with the license change too. 
I really did not change much anyways

lanistor, 2022-02-08:
@Sam-Izdat Hi, may we get your opinion?

martinberlin, 2022-02-08:
@mmib and @SimonSelg are the last two collaborators in my list. Do you agree to the License update?

vroland, 2022-02-13:
I think this is complete now, since the weatcher example contributed by @mmib is under the copyright of the original author
anyway and not covered by the GPL3, and @SimonSelg contributed to hardwar production files.

martinberlin, 2022-02-13:
Great so then it could be updated to LGPL right?
Looking forward to see if at some point my take to use EPDiy also for LVGL could be accepted. Thanks!

commit f3feb5a9 (2022-02-13): change driver license to LGPL3 (fix #138)
closed 2022-02-13 by vroland