vroland/epdiy #139: Reason for op-amp buffer on Vcommon

dot4qu, 2022-01-13:
Hi!

Would you mind letting me know why you sent the voltage divider output for Vcommon through an op amp? I'm assuming it's to show a low impedance to the screen, but the data sheet for the ED060SC4 shows that the max current draw on the Vcommon pin is 0.2mA. The +/-15 for the source voltage pins draws much more at 36mA max.

What was your reasoning for using an op amp for Vcommon and not for any other voltage inputs? No judgement here I'm just curious - I'm trying to figure out when to buffer voltages in my own circuits like this. Thanks!

vroland, 2022-01-14:
I'ts mainly to ensure that VCOM really stays on-point. On older revisions, there where visible artifacts when not adding an additional capacitor on VCOM, so there seem to be some current spikes. So overall, I wanted to just be sure, since the displays are very sensitive to VCOM being correct.

dot4qu, 2022-01-16:
Interesting, so not what I was expecting. Thanks!

closed 2022-01-16 by dot4qu