vroland/epdiy #142: Slack invitation link in badge has expired

sychan, 2022-01-21:
I went to try the Slack workspace, but the link has expired. Is the Slack workspace still going? Maybe update or remove the slack badge?

martinberlin, 2022-01-21:
Hello Steve, I just sent you an email invitation to Slack with the address you have in the GitHub profile.

mickeprag, 2022-01-21:
I would like to join as well. Better perhaps to fix the invite badge?

martinberlin, 2022-01-21:
If you want Micke I can send you an invitation to an email address. Only @vroland I suppose has the rights to fix the Readme page and to get a new link from Slack.

mickeprag, 2022-01-21:
Yes, please send me a link as well. Thank you.

martinberlin, 2022-01-21:
> Yes, please send me a link as well. Thank you.

Sorry to what email ? (There is none in your GitHub profile)

mickeprag, 2022-01-21:
micke.prag@gmail.com

vroland, 2022-01-21:
Unfortunately, the invite link expires every 30 days and slack does not allow to extend this period :/ So thanks for the reminder, I updated the link :)

commit f35c2822 (2022-01-21): Update slack invite link (fix #142)
closed 2022-01-21 by vroland