vroland/epdiy #143: Touch screen support

Pickaxe-Hit, 2022-01-25:
Hello!
I found the module "ED060KD1" in the README file, and this screen has a touch controller(using IIC).
I want to know if epdit support the touch fonction?
Thanks.

martinberlin, 2022-01-25:
Hi there,
AFAIK currently EPDiy is only a display driver and does not support any touch interface. The only display that comes with touch so far and runs an old version of epdiy is the 4.7" Lilygo EPD47.
I think you will have to code the touch component yourself unless you can find a compatible and working C driver. 
I did it only for ESP-IDF with the L58 driver for Lilygo, you can use this as a [base for yours](https://github.com/martinberlin/FT6X36-IDF). That said, I think this issue can be closed, since it's not the intention of EPDiy to support also touch, or at least not that I'm aware of.

Pickaxe-Hit, 2022-01-25:
I found a new screen which is made by AUO. The biggest advantage is "cheap". It is only 48CNY or about 7$. I found it at a Chinese trading platform. Although the connector is quite different (50-pins FPC), but the pin function is similar(8-bit Parallel Data). Can epdiy drive the screen?(I can design the hardware but I'm not good at the software) I will put the datasheet here. Thanks!
[datasheet.pdf](https://github.com/vroland/epdiy/files/7935066/datasheet.pdf)

Pickaxe-Hit, 2022-01-26:
> 我发现了一个友达制造的新屏幕。最大的优势是“便宜”。只需 48 元人民币或约 7 美元。我是在中国的交易平台上找到的。虽然连接器有很大不同（50-pins FPC），但引脚功能相似（8-bit Parallel Data）。epdiy可以驱动屏幕吗？（我可以设计硬件但我不擅长软件）我将数据表放在这里。谢谢！ [数据表.pdf](https://github.com/vroland/epdiy/files/7935066/datasheet.pdf)

@martinberlin

Pickaxe-Hit, 2022-01-26:
> 我真的不知道它是否能够控制那个屏幕。请在此存储库的自述文件中查看支持的显示。 但无论如何，如果你想控制它，你将不得不制作一个新的 PCB。那么为什么不坚持使用受支持的屏幕之一呢？ 而@Pickaxe-Hit这张票是关于别的东西的，所以你正在将上下文更改为不同的东西，这个问题应该更适合其他渠道，如 epdiy.slack.com

That because I want to use a quite cheap module. I'm sorry for applying a wrong issue. I will close it at once.

closed 2022-01-26 by Pickaxe-Hit