vroland/epdiy #146: LilyGo T5 4.7inch - eInk update latency

ashald, 2022-01-31:
LilyGo T5 4.7inch claims full frame redraw in ~630ms, yet using code based on https://github.com/vroland/epdiy/pull/79/files and doing:
```
EpdRect some_rect = {
    .x = 0,
    .y = 0,
    .width = width,
    .height = height
  };
epd_fill_rect(some_rect, c, fb);
err = epd_hl_update_screen(&hl, MODE_GC16, this->temperature_); // This takes ~1.5s
epd_poweroff();
epd_deinit();
```
but it takes about 1.5s to do that.
Also, I have to add some delay before `epd_poweroff()` or otherwise image beings to fade immediately.

Are there any ways to make it faster? Is it bound by CPU, SPI throughput or something else?..

I tried other modes like `MODE_GC16_FAST` but i doesn't seem to work - it t just doesn't do anything. I had the confits parameters set in the ini as in the PR but to no avail.

I only need to draw black & white - could you suggest how can I speedup the above?
Also, is there a way to avoid need for another extra delay before `epd_poweroff()`?

I'm working on a battery-powered setup that uses BLE to received the data every minute and render it on the screen. So far, it takes 2s to boot up and fetch data, and almost 3 to update the screen. I was hoping I could bring it down to 1s.

Thanks!

martinberlin, 2022-01-31:
Hi Ashald,
With the new Waveform that recently Lilygo open sourced, it's slower but black is more black. If you want it faster try the old default waveform:

https://github.com/martinberlin/epdiy-rotation/blob/master/waveforms/epdiy_ED047TC1.h

>I tried other modes like MODE_GC16_FAST but i doesn't seem to work - it t just doesn't do anything. I had the confits parameters set in the ini as in the PR but to no avail.

This Mode as far as I know is not available for this display. Use MODE_DU to go faster specially if you need only black/white.

> Are there any ways to make it faster? Is it bound by CPU, SPI throughput or something else?

Not that I'm aware off, if ESP32 CPU is at 240 Mhz, that is as fas as can go. There is no SPI except for external RAM, this epapers use parallel communication.

The fading of this display is not the fault of EPDiy but a hardware issue with the Lilygo. With other epapers this does not happen. If you use battery it will probably happen only when there is no more power. Everytime you leave this EPD somewhere for a long time, you find it black.

martinberlin, 2022-01-31:
@vroland in the last update on the Lilygo  Waveform I think there was an issue:
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/master/Waveform_header

The Waveform that they left in the official repository is ED047TC2.h , but it was submitted here and received the name ED047TC1.h overriding old Waveform that still worked (And is faster than TC2)
In my opinion the two should be left there, being ED047TC2 the default, so at least you have the option to use the old one that is faster.

vroland, 2022-01-31:
@martinberlin Yes, you are right, we should probably include both the epdiy waveform and the vendor waveform.
That is probably the main issue, because the vendor waveform uses a lot more cycles than the epdiy waveform.

However, maybe a similar issue I had with the JPEGDEC example occurs here?
@ashald Are you using C++ or pure C in your code? And are you using the ESP-IDF or some other platform?
Because there may be some issue where update is only half as fast in some cases, but I couldn't pin down the reason so far...

ashald, 2022-01-31:
Thanks for quick replies!

@martinberlin, just tried to use it, but getting a compilation error `sorry, unimplemented: non-trivial designated initializers not supported`. I'm sorry my C/C++ knowledge is very limited, is there anything you can suggest to deal with this error?

I tried using `MODE_DU`, it improves update time from ~1.5s to ~1s, but it leaves significant ghosting when it goes from black into white. I'd imagine I could try clearing the display first while data is being prepared, and then using `MODE_DU` to draw new text.

Also, since you mentioned issues with Lilygo design, are there any eInk screens you can recommend that offer 4"+ size and sub-second monochrome updates? In case you know any...

@vroland I've been using a custom component for ESPHome, which is C++, but I'm not sure I can answer the second part of your question. It seems the default configuration is to use Arduino platform, but then I saw some functions from ESP-IDF. 🤯

martinberlin, 2022-01-31:
> sorry, unimplemented: non-trivial designated initializers not supported

No idea because I don't get this. For sure you are compiling this with C++ somewhere since "the order of initialization needs to be in the exact order of declaration." otherwise gives this error (I think) So check if the structures there defined have any of the parameters in the wrong order. This does not happens if you compile a C example with ESP-IDF

ashald, 2022-02-05:
Thanks a lot!

And one more somewhat related question of you don't mind.
I'm still trying to make my Lilygo work before giving up on it, and one thing I noticed is that with deep sleep and `MODE_GC16` some areas do not refresh and quickly accumulate some garbage. What helps with that is doing this - basically drawing entire screen black, and then drawing it white with a padding of 1 pixel.
```
epd_fill_rect({ .x=0, .y=0, .width=EPD_WIDTH,   .height=EPD_HEIGHT  }, 0x00, fb);
epd_fill_rect({ .x=1, .y=1, .width=EPD_WIDTH-1, .height=EPD_HEIGHT-1}, 0xFF, fb);
```
With the above, `MODE_GC16` gives a nice crisp picture after refresh, but executing above 2 lines takes almost 400ms. Is there a way I can achieve a similar effect in a more efficient way?

I saw that `epd_hl_set_all_white` uses `memset(state->front_fb, 0xFF, fb_size);` which executes much faster, but I cannot figure out how to use it with the "pixel packing" and padding.

Thanks!

martinberlin, 2022-02-05:
@ashald you can of course speed it up using a memset directly on the framebuffer that set's all to 0x00. But you will have to still send the other like this, unless you figure out how to make a for ...loop for every row to send the bytes shifted.

martinberlin, 2022-02-26:
> we should probably include both the epdiy waveform and the vendor waveform. That is probably the main issue, because the vendor waveform uses a lot more cycles than the epdiy waveform. we should probably include both the epdiy waveform and the vendor waveform.

Shall I make the pull request adding both Waveforms?
Then I guess this issue can be closed.

martinberlin, 2022-03-02:
@ashald if there is nothing more to be discussed in this issue please close it. Will make a pull request to add both waveforms later but probably in a dedicated issue, not related to this one.

closed 2022-03-03 by ashald