vroland/epdiy #148: Make this a PlatformIO Library

matteocarnelos, 2022-02-02:
It would be awesome for this library to be a PlatformIO Library, available in the PlatformIO Registry.

It should be enough to properly fill the `library.json` file as described in [Creating Library - PlatformIO](https://docs.platformio.org/en/latest/librarymanager/creating.html).

vroland, 2022-02-06:
Right, that would probably make sense. It would need some testing though. And it would be nice if it could be used seamlessly
with either the arduino framework or esp-idf through platformio. I don't have the time to work that all out right now, but I'm happy to test if someone creates a suitable metadata file.

martinberlin, 2022-02-26:
@matteocarnelos after the merge is there anything more to be tested here or we can close this issue?

matteocarnelos, 2022-03-01:
I think everything's done and this issue has been solved

closed 2022-03-01 by matteocarnelos