vroland/epdiy #150: a  strange error in show String in LilyGo T5 4.7inch

lixy123, 2022-02-04:
I use lilygo T5 4.7inch - eink. 
When testing the display text in eink, I found a strange error at show text.

After the second screen clearing,  if the displayed text content and position are exactly the text of the previous screen, it will be blank show.  

Here is demo code rewritten by me from / Master / examples / lilygo-t5-47-epd-platform,
run at lilygo T5 4.7inch - eink.
"Hello to you 1 \ nholo to you 2" should be displayed when the the second time looped, but I found that "hello to you 1" does not display, Show only "hello to you 2"...

It is a bug? or writing skills are needed?

#include <Arduino.h>
#include "epd_driver.h"
#include "epd_highlevel.h"
#include "Firasans.h"

#define WAVEFORM EPD_BUILTIN_WAVEFORM

EpdiyHighlevelState hl;
// ambient temperature around device
int temperature = 20;
uint8_t *fb;
enum EpdDrawError err;

int show_cnt;

void display_message_test(int cursor_x, int cursor_y, char * text) {

  epd_poweron();
  epd_clear();
  epd_poweroff();

  // first set full screen to white
  epd_hl_set_all_white(&hl);

  EpdFontProperties font_props = epd_font_properties_default();
  font_props.flags = EPD_DRAW_ALIGN_LEFT;

  epd_write_string(&FiraSans_12, text, &cursor_x, &cursor_y, fb, &font_props);

  epd_poweron();
  //epd_clear();
  err = epd_hl_update_screen(&hl, MODE_GC16, temperature);
  epd_poweroff();
  delay(1000);
}

void setup() {
  Serial.begin(115200);

  epd_init(EPD_OPTIONS_DEFAULT);
  hl = epd_hl_init(WAVEFORM);
  fb = epd_hl_get_framebuffer(&hl);
  //epd_set_rotation(orientation);

  show_cnt = 0;
  epd_clear();
}

void loop()
{
  if (show_cnt>=3) return; 
  
  show_cnt = show_cnt + 1;
  Serial.println("show " + String(show_cnt) + " time");
  if (show_cnt == 1)
  {
    display_message_test(20, 50, "Hello to you 1");  
  }
  else if (show_cnt == 2)
  {
    display_message_test(20, 50, "Hello to you 1\nHello to you 2");
  }
  else  if(show_cnt == 3)
  {
    display_message_test(20, 50, "Hello to you 1\nHello to you 2\nHello to you 3");
  }

  delay(5000);
}

lixy123, 2022-02-04:
Add one sentence:
If the text is displayed after call deep_sleep, there is no such phenomenon.
Just like the code example here  https://github.com/vroland/epdiy/tree/master/examples/lilygo-t5-47-epd-platformio

martinberlin, 2022-02-04:
This does not seem like a but to me. 
If you do a :

```c
void loop() {
  if (show_cnt>=3) return;
  //...
```

if your code, it will of course return from the loop and do nothing when the counter arrives to nr. 3. Change it to 4 or remove the "=" and you will get the expected result.

lixy123, 2022-02-05:
The setup function has this code: "show_cnt = 0;"
the first loop ,should show "Hello to you 1"
the 2st loop, should  show "Hello to you 1\nHello to you 2"
the 3st loop, should  show "Hello to you 1\nHello to you 2\nHello to you 3"

but, Not so, some String is Eaten by the display screen， 
haha...

martinberlin, 2022-02-05:
Did you tried what I recommended? This is not a bug from EPDiy

martinberlin, 2022-02-05:
Tested this with following code compiling it with idf.py flash in the examples demo directory as test.c:

```c
#include "esp_heap_caps.h"
#include "esp_log.h"
#include "esp_timer.h"
#include "esp_types.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "epd_driver.h"
#include "epd_highlevel.h"
#include "firasans_12.h"

#define WAVEFORM EPD_BUILTIN_WAVEFORM
EpdiyHighlevelState hl;
int temperature = 20;
uint8_t *fb;
enum EpdDrawError err;
int show_cnt;

void delay(uint32_t millis) { vTaskDelay(millis / portTICK_PERIOD_MS); }

void display_message_test(int cursor_x, int cursor_y, char * text) {
  epd_poweron();
  epd_clear();
  epd_poweroff();
  epd_hl_set_all_white(&hl);
  EpdFontProperties font_props = epd_font_properties_default();
  font_props.flags = EPD_DRAW_ALIGN_LEFT;
  epd_write_string(&FiraSans_12, text, &cursor_x, &cursor_y, fb, &font_props);
  epd_poweron();
  err = epd_hl_update_screen(&hl, MODE_GC16, temperature);
  epd_poweroff();
  delay(1000);
}

void loop()
{
if (show_cnt>=3) return;

show_cnt = show_cnt + 1;
printf("show %d time\n",show_cnt);
if (show_cnt == 1)
{
display_message_test(20, 50, "Hello to you 1");
}
else if (show_cnt == 2)
{
display_message_test(20, 50, "Hello to you 1\nHello to you 2");
}
else if(show_cnt == 3)
{
display_message_test(20, 50, "Hello to you 1\nHello to you 2\nHello to you 3");
}
delay(500);
}

void app_main() {
  epd_init(EPD_OPTIONS_DEFAULT);
  hl = epd_hl_init(WAVEFORM);
  fb = epd_hl_get_framebuffer(&hl);
  show_cnt = 0;
  epd_clear();
  while (true) {
    loop();
  }
}
```
It works as expected and prints the 3 lines in the EPD.

lixy123, 2022-02-06:
It is normal.

I understand that in order to maximize and fully improve the efficiency of ink screen display, intelligent display block technology is used.

Although through the use of code of “EPD_ Fullclear”  solves the problem of display, but it feels strange to understand. It is estimated that many novices who use this API for the first time will be confused when they encounter this problem. It is recommended to correct if it is a design bug, or write this phenomenon in the introduction.

Thank you for your reply!

martinberlin, 2022-02-08:
No problem @lixy123 
You can close here if your problem is solved then.  I will ask the maintainer if this could be added, so more people understand correctly how the updates to the display works.
In my opinion, this is not a design bug, it was designed like this.
Thanks a lot for your recommendations.

closed 2022-02-17 by lixy123