vroland/epdiy #152: What's the fast way to update screen, while having a fullscreen's color buffer?

lanistor, 2022-02-08:
We are using lvgl as UI library, for rendering on epaper, we set each pixel's color by EpdiyHighlevelState, then using `epd_hl_update_area` to render.
But `epd_hl_update_area` will cost nearly 3 seconds, for a fullscreen rendering (display size: 1024x758).
[Demo code](https://github.com/martinberlin/lvgl_epaper_drivers/blob/master/lvgl_tft/epdiy_epaper.c)

Is this the best way? Is there a faster way to do it?

vroland, 2022-02-09:
Hm, 3 seconds seems really long for epd_hl_update_area() at this resolution, especially with the DU waveform. Is your PSRAM set to 80MHz and QIO?

lanistor, 2022-02-09:
Thanks for the advice, just changed my configuration to: 
```
Flash SPI mode: QIO
RAM clock speed: 80MHz
```
and without finding `PSRAM  QIO` option.

After this change, the time reduce to 2.5s. Have improved, but still too long.

And, i'm using `MODE_GC16` mode, not `MODE_DU`.

vroland, 2022-02-13:
Just to see if that changes anything: Could you try disabling "try to allocate memories of wifi...",
"enable workaround for bug in..." and "enable bank switching"?

martinberlin, 2022-02-15:
I would also advice to use MODE_DU and follow point 9 of this:
https://github.com/martinberlin/lv_port_esp32-epaper#build-and-run-the-demo
the 16 grayscale mode imho is simply not fast enoughto render UX in Eink with EPDiy (maybe with a smaller resolution like 800x600 could work) but I would not recommend it. You need to decide if you want speed or nice 16 grays, I would choose first option.

lanistor, 2022-02-18:
> 

This works, after disabling "enable workaround for bug in..." and "enable bank switching", the time reduce to 2000ms, disabling "try to allocate memories of wifi..." will cause crash, the memory error: "E (5811) epd: could not allocate LUT!".

But, 2000ms still seems too long.

lanistor, 2022-02-18:
> I would also advice to use MODE_DU and follow point 9 of this: https://github.com/martinberlin/lv_port_esp32-epaper#build-and-run-the-demo the 16 grayscale mode imho is simply not fast enough to render UX in Eink with EPDiy (maybe with a smaller resolution like 800x600 could work) but I would not recommend it. You need to decide if you want speed or nice 16 grays, I would choose first option. I‘ve made a PR with my own lvgl driver for EPDiy but was never merged due to a license conflict.

I think in a real project, different greys are usually needed.

vroland, 2022-02-27:
With better hardware there sure is. The ESP32 is just not that fast however. When you consider its memory bandwidth 
it is already a lot slower than the display could theoretically accept... So if you want the fastest updates possible, 
you probably need a commercial controller or FPGA...

closed 2022-05-05 by martinberlin