vroland/epdiy #153: U3 and U7 missing from Pick&Place file for V6 gerbers

errolt, 2022-02-08:
Looking at epaper-breakout-top-pos.csv, U3(TPS651851) and U7(TPS2113APW) are missing.
Not an issue for me, but might be for anyone that sends away to get manufactured.

vroland, 2022-02-13:
Good to know, I'll have a look if i re-generate these files at some point. For now, they don't seem to be available for pick & place anyway.

martinberlin, 2022-05-14:
Shall I add this in the Pick & Place @errolt ?  Or can you do it yourself, if you are also a KiCad user, and make a pull request?

martinberlin, 2022-05-15:
@errolt please check the last commit in my fork. Just added this 2 components.
If that is OK then I will make a pull request and we add them in master.

Note that this two components are not present in the all-pos CSV but they exist in the [JLCPCB position file epaper-breakout-top-jlcpcb.csv](https://github.com/vroland/epdiy/blob/master/hardware/epaper-breakout/gerbers/epaper-breakout-top-jlcpcb.csv)

errolt, 2022-05-19:
@martinberlin Sorry for the late reply. Thank you for adding them, and **it looks good**. 
I just know enought about Kicad to open a schematic, not enough to change anything.

closed 2022-06-02 by martinberlin