vroland/epdiy #156: Epdiy not compatible with ESP-IDF v5

jdoubleu, 2022-02-14:
I noticed, that this project is not compatible with the upcoming ESP-IDF version 5. To be fair, that version is still in _early_ development and looking at their [support periods](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#support-periods) can be estimated to be released mid to end 2022.

The migration guide can be found here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/index.html.

Getting the latest ESP-IDF environment and compiling the [demo](https://github.com/vroland/epdiy/tree/master/examples/demo) can be done with the [`espressif/idf:latest`](https://hub.docker.com/r/espressif/idf) docker image:
```sh
git clone --recursive https://github.com/vroland/epdiy.git
docker run --rm -it -v epdiy:/app espressif/idf:latest sh -c 'cd /app/examples/demo; idf.py build'
```

<details> 
  <summary>Build log</summary>

```
[877/898] Building C object esp-idf/epd_driver/CMakeFiles/__idf_epd_driver.dir/display_ops.c.obj
FAILED: esp-idf/epd_driver/CMakeFiles/__idf_epd_driver.dir/display_ops.c.obj 
ccache /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -Iconfig -I/app/src/epd_driver/include -I/opt/esp/idf/components/newlib/platform_include -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/include -I/opt/esp/idf/components/freertos/esp_additions/include/freertos -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/opt/esp/idf/components/freertos/esp_additions/include -I/opt/esp/idf/components/esp_hw_support/include -I/opt/esp/idf/components/esp_hw_support/include/soc -I/opt/esp/idf/components/esp_hw_support/include/soc/esp32 -I/opt/esp/idf/components/esp_hw_support/port/esp32/. -I/opt/esp/idf/components/heap/include -I/opt/esp/idf/components/log/include -I/opt/esp/idf/components/lwip/include/apps -I/opt/esp/idf/components/lwip/include/apps/sntp -I/opt/esp/idf/components/lwip/lwip/src/include -I/opt/esp/idf/components/lwip/port/esp32/include -I/opt/esp/idf/components/lwip/port/esp32/include/arch -I/opt/esp/idf/components/soc/include -I/opt/esp/idf/components/soc/esp32/. -I/opt/esp/idf/components/soc/esp32/include -I/opt/esp/idf/components/hal/esp32/include -I/opt/esp/idf/components/hal/include -I/opt/esp/idf/components/hal/platform_port/include -I/opt/esp/idf/components/esp_rom/include -I/opt/esp/idf/components/esp_rom/include/esp32 -I/opt/esp/idf/components/esp_rom/esp32 -I/opt/esp/idf/components/esp_common/include -I/opt/esp/idf/components/esp_system/include -I/opt/esp/idf/components/esp_system/port/soc -I/opt/esp/idf/components/esp_system/port/include/private -I/opt/esp/idf/components/xtensa/include -I/opt/esp/idf/components/xtensa/esp32/include -I/opt/esp/idf/components/driver/include -I/opt/esp/idf/components/driver/esp32/include -I/opt/esp/idf/components/driver/deprecated -I/opt/esp/idf/components/esp_pm/include -I/opt/esp/idf/components/esp_ringbuf/include -I/opt/esp/idf/components/efuse/include -I/opt/esp/idf/components/efuse/esp32/include -I/opt/esp/idf/components/vfs/include -I/opt/esp/idf/components/esp_wifi/include -I/opt/esp/idf/components/esp_event/include -I/opt/esp/idf/components/esp_netif/include -I/opt/esp/idf/components/esp_eth/include -I/opt/esp/idf/components/tcpip_adapter/include -I/opt/esp/idf/components/esp_phy/include -I/opt/esp/idf/components/esp_phy/esp32/include -I/opt/esp/idf/components/esp_timer/include -I/opt/esp/idf/components/esp_adc_cal/include -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -O2 -fmacro-prefix-map=/app/examples/demo=. -fmacro-prefix-map=/opt/esp/idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-1554-g20847eeb96\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/epd_driver/CMakeFiles/__idf_epd_driver.dir/display_ops.c.obj -MF esp-idf/epd_driver/CMakeFiles/__idf_epd_driver.dir/display_ops.c.obj.d -o esp-idf/epd_driver/CMakeFiles/__idf_epd_driver.dir/display_ops.c.obj -c /app/src/epd_driver/display_ops.c
/app/src/epd_driver/display_ops.c: In function 'fast_gpio_set_hi':
/app/src/epd_driver/display_ops.c:32:3: error: 'GPIO' undeclared (first use in this function)
   GPIO.out_w1ts = (1 << gpio_num);
   ^~~~
/app/src/epd_driver/display_ops.c:32:3: note: each undeclared identifier is reported only once for each function it appears in
/app/src/epd_driver/display_ops.c: In function 'fast_gpio_set_lo':
/app/src/epd_driver/display_ops.c:36:3: error: 'GPIO' undeclared (first use in this function)
   GPIO.out_w1tc = (1 << gpio_num);
   ^~~~
/app/src/epd_driver/display_ops.c: In function 'epd_base_init':
/app/src/epd_driver/display_ops.c:77:3: error: implicit declaration of function 'PIN_FUNC_SELECT' [-Werror=implicit-function-declaration]
   PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[CFG_DATA], PIN_FUNC_GPIO);
   ^~~~~~~~~~~~~~~
/app/src/epd_driver/display_ops.c:77:19: error: 'GPIO_PIN_MUX_REG' undeclared (first use in this function); did you mean 'GPIO_PIN_REG_31'?
   PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[CFG_DATA], PIN_FUNC_GPIO);
                   ^~~~~~~~~~~~~~~~
                   GPIO_PIN_REG_31
/app/src/epd_driver/display_ops.c:77:47: error: 'PIN_FUNC_GPIO' undeclared (first use in this function)
   PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[CFG_DATA], PIN_FUNC_GPIO);
                                               ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
[879/898] Linking C static library esp-idf/wifi_provisioning/libwifi_provisioning.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
</details>

This issue tries to list all known incompatibilities so far:
1. [ ] It looks like some `GPIO` functions and macros are not available anymore (used in [`display_ops.c`](https://github.com/vroland/epdiy/blob/master/src/epd_driver/display_ops.c#L36)). The migration guide only mentions the [removal of a configuration](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/peripherals.html#gpio) and the `rtc_gpio_desc` object. At least I [found some of the macros](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/io_mux_reg.h#L93) still in the codebase, so maybe they've just been moved in the headers.
2. [ ] Not included in the above build log, but when you include/compile [`pca9555.c`](https://github.com/vroland/epdiy/blob/master/src/epd_driver/pca9555.c), you'll see the following error message:
   ```
   /app/epdiy/src/epd_driver/pca9555.c: In function 'i2c_master_read_slave':
   /app/epdiy/src/epd_driver/pca9555.c:33:63: error: 'portTICK_RATE_MS' undeclared (first use in this function); did you mean 'portTICK_PERIOD_MS'?
   esp_err_t ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
                                                               ^~~~~~~~~~~~~~~~
                                                               portTICK_PERIOD_MS
   ```
   Indeed it [looks like](https://github.com/espressif/esp-idf/blob/master/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h#L1028) the macro `portTICK_RATE_MS` was already deprecated and they just [disabled the backwards compatible switch's](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/freertos.html#legacy-api-and-data-types) default. Luckily, it can just be replaced by the `portTICK_PERIOD_MS` macro (see https://github.com/espressif/esp-idf/issues/51).
3. _TBC..._

martinberlin, 2022-06-02:
This looks good @jdoubleu !
On hold till #183 is merged.

martinberlin, 2022-12-02:
Wanted to comment that IDF version 5.0 was released today (Sorry I'm 4 hours late)

PR #180 (2023-01-17): get ready for ESP-IDF v5
Fixes #156

WIP: Pleas do not merge, yet!

~At the moment, this PR is based on works in #177, so the CI is running.~ CI doesn't run, because the GitHub CI config has not been merged into `master`, yet.

## Issues
Some symbols were no longer implicitly imported, so I included some rather specific headers.

However, there are a few issues remaining:
1. Some functions, like `rtc_clk_apll_enable` changed their signature. They are currently used in [`i2s_data_bus.c`](https://github.com/vroland/epdiy/blob/master/src/epd_driver/i2s_data_bus.c#L191).
2. The `RMTMEM` global var [... cut]
closed 2023-01-17 by vroland