vroland/epdiy #159: LilyGo T5 4.7inch - eInk  display effect is darker when use library of epdiy,   Compared with lilygo's library

lixy123, 2022-02-18:
I use the official sample demo example of lilygo T5 47 ink screen library and the epdiy library demo example code, 
and found that the example code in epdiy is slightly dark,  I feel that I didn't do it right everywhere

You can compare  run result of  the two example code Light hearted:
https://github.com/vroland/epdiy/tree/master/examples/lilygo-t5-47-epd-platformio
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/master/examples/demo

It's dimmed a lot. 
The display effect makes people look unhappy. Can you give me some suggestions

vroland, 2022-02-18:
Could you check if this PR fixes that? https://github.com/vroland/epdiy/pull/158

martinberlin, 2022-02-26:
@lixy123 can you please pull the latest master and try out?
I cannot reproduce this, can you please drop some photos as a response to show this effect you are seeing?

martinberlin, 2022-04-22:
Hi @lixy123 can you please give this a try?
When more than two months pass from the request from feedback I suggest we consider this as a stale Issue and close. Please respond if you want to keep this alive.

martinberlin, 2022-04-23:
Closing here since the same issue is being addressed on https://github.com/vroland/epdiy/issues/171
And it has been also almost 2 months since last request of feedback. Please follow latest pull request for updates since I am quite sure this is related to the new waveform

closed 2022-04-23 by martinberlin