vroland/epdiy #161: M5Paper in Arduino

Yardie-, 2022-02-23:
Hi I have used your library on a Lilygo and would now like to use it on an M5paper.
There are references to people doing this but it's not totally clear how.
Can the specifically required defines be setup prior to including the library.
Can someone point me in the right direction.

Thanks.
Greg

martinberlin, 2022-02-23:
@Yardie- even tought there might be some developers using the epdiy functions to draw the buffer and then send it to any epaper (For ex. this [Epub-Reader project](https://github.com/atomic14/diy-esp32-epub-reader/blob/main/src/boards/M5Paper.h))
This component has really nothing to do with M5 epaper since that one uses IT8951 as an epaper controller. EPDiy is the epaper controller and it needs compatible hardware like Lilygo or one of the boards you can see in the [Hardware section](https://github.com/vroland/epdiy/tree/master/hardware).
Feel to close this one if there is nothing more to add.

Yardie-, 2022-02-23:
@martinberlin thanks that seems to be the obvious solution.
I was looking at the epub stuff and was thinking it may be the way to go.

Thanks I wasn't clear on the separation. 

I do like the way EPDiy works with the buffer especially handling fonts.
I would love to add the down scaling I worked out for 1bpp waveshare. 
But maybe later.
I just want to get some stuff working with the hardware I have.
Thank you for your time.

martinberlin, 2022-02-23:
Greg,
one more thing, with M5 epaper you can also use:
https://github.com/lovyan03/LovyanGFX

I don't have M5, but I do have some other Waveshare epapers that are controlled with IT8951 and I find that component great.
Here you have an example that I did to render a JPG using that library
https://github.com/martinberlin/cale-idf/blob/feature/45-it8951/main/demos/demo-it8951-lovyan.cpp

Yardie-, 2022-02-23:
Thanks @martinberlin.
I have an old M5 that I never got arround to doing much with due to the whole load the whole ttf thing. Now the battery is stuffed but it is still a slick unit. (don't have a 3d printer.
I was really thinking of going back to the goodisplay 2.7. I looked at elcal ages ago. :)
I have stuffed around with this for years. Now it's time to get some of these epapers I have  actually doing useful things. 
I really enjoy being able to just include a few the characters from one ttf at one size. 
But also the version in the LilyGo official doesn't work as well as the EPDiy rendering. Except for sleep. The LilyGo official has a function power_off_all which turns the led of in deep sleep. That doesn't work in the vroland version. Even when basically did the same calls. I clearly missed something.

I will take your advice and give the LovyanGFX lib a go. 
Thanks.

Yardie-, 2022-02-23:
WOW thanks heap Martin (@martinberlin ) 
that's a great library.