vroland/epdiy #162: LilyGo turns off touch with poweroff when battery powered.

Yardie-, 2022-02-24:
Hi guys I have been running around in circles thinking my LilyGo was broken
but I found the problem.
the lines in config_reg_vh2.h 
```
static void cfg_poweroff(epd_config_register_t *cfg) {
#if defined(CONFIG_EPD_BOARD_REVISION_LILYGO_T5_47)
  // This was re-purposed as power enable.
  cfg->ep_scan_direction = false;
#endif
```
are turning the epd off but also the touch interface when the unit is battery powered.
When this is set to true the touch works fine.

is the a simple way within your framework to change this behaviour 

the way the lilygo code handles this is to have two functions
epd_poweroff 
and epd_poweroff_all
one leaves the value at true the other to false

I can think of several solutions and I can hack out a simple local fix 
but it's a simple fix which would probably be best solved by those in the loop of how you want to do this.
Thanks.
Greg

Yardie-, 2022-02-24:
Actually setting the value to true allows the touch to stay active and the blue led goes out when you send the esp into deep sleep but still wakes on touch.
I haven't done a power check but the touch interface is obviously drawing something still being on.

martinberlin, 2022-02-24:
Hi @Yardie-
I will take a look at this. That means that you could theoritically use the Touch INT pin to wake up from deepsleep?

I was trying this but I failed and now I know why. If you cannot make a PR just tell me and I will do it.

@vroland it would be alright to pass a parameter to epd_poweroff + cfg_poweroff for the displays that have touch?
Or you can imagine a better way to do it for Lilygo model?

Yardie-, 2022-02-24:
Thanks @martinberlin 
What I thought might be a nice trick is to put two functions inside the define
ie  
#if defined(CONFIG_EPD_BOARD_REVISION_LILYGO_T5_47)

continue code with set as true 
then end that function then 
still within the define 
add another function
cfg_poweroff_all ....
and set that  function with the value as false and end the define.
as it does already 

then in the caller just add a section

```
#if defined(CONFIG_EPD_BOARD_REVISION_LILYGO_T5_47)

void epd_poweroff_all() {
  cfg_poweroff_all(&config_reg);
  i2s_gpio_detach();
}
#endif

```
with the same in epd_driver.h

then the non lilygo user doesn't even have the function and nothing has changed

I am a little clueless on PRs and GIT
So I would love to handball that to you. :)

Thanks again Martin

BTW 
the i2c issue I had was a dodgey RTC.
ESP32's are also quite picky with i2c so the pullups have to be sorted properly.
Anyway that's another story.

Yardie-, 2022-02-24:
Ok I couldn't sleep so I implemented it that way and it seems to work fine.
Now power can be regularly be turned off the epd without affecting the touch then to sleep I did the  following.
The LilyGo will still wake on touch (possibly because of the capacitance change of the touch is picked up by the ESP. But I can't really test that )
```
   esp_sleep_enable_ext1_wakeup(GPIO_SEL_13, ESP_EXT1_WAKEUP_ANY_HIGH);
    epd_poweroff_all();
    epd_deinit();
    esp_deep_sleep_start();
```
Code changes at
https://github.com/Yardie-/epdiy/blob/master/src/epd_driver/config_reg_v2.h
https://github.com/Yardie-/epdiy/blob/master/src/epd_driver/display_ops.c
https://github.com/Yardie-/epdiy/blob/master/src/epd_driver/include/epd_driver.h

I don't know how to do a PR.

Have fun.
Greg

martinberlin, 2022-02-24:
Greg checkout somewhere this:
https://github.com/martinberlin/epdiy-rotation/tree/vroland-master

And try my update. I won't propose any pull request until the maintainer gives his feedback and that can last until beginning of March. Not sure if this is the best way to do it. 
I'm not going to duplicate code doing a **cfg_poweroff_all** copy of cfg_poweroff just for the Lilygo exception. I propose to set an additional parameter that can be left empty, or set to true in epd_poweroff()   but leaving it empty will not affect other boards.

Yardie-, 2022-02-24:
@martinberlin Thanks Martin yes the is totally appropriate this was really just a proof of concept thing done at 1 in the morning on a school night. (dad joke)
Which ever way you go guys go I am sure will be the a correct path for the code. 

Thanks.
Greg

Yardie-, 2022-02-25:
Sorry @martinberlin Arduino doesn't like having two different definitions.
One in 
epd_driver.h as  void epd_poweroff();
and another in 
display_ops.h as void epd_poweroff(bool * poweroff_all);

I'll stick with my code for now and at leave it up to you guys to work out.
I know my code ain't pretty but it works ok for now and doesn't change the API

:)
Thanks heaps for your time.

martinberlin, 2022-02-25:
Sorry I could not reproduce it. So I guess my code is also crap ;)
Usually you could have two definitions of the same function that accept different parameters.

Yardie-, 2022-02-25:
@martinberlin :)
AFAIK overriding only works in cpp but in c I didn't think so.
The compiler of Arduino may be playing a little tougher than the ESP compiler.
I think that maybe a combination of both ideas is probably the go.
So only lilygo users have the poweroff_all function and the definition of the 
cfg_poweroff(&config_reg);
could be simply wrapped in another define.

It's really just a swing a cat situation.

We have a solution that works. 
It just needs to be refined.

Did you try the wake on touch thing.
It surprises me a little if the touch screen is actually turned off that it would send the capacitance down the int wire. But hey I'm not an electrical engineer.
I let way too much smoke out to consider myself  one of those. 
:)

vroland, 2022-02-27:
Hm, it's really annoying that they just re-used this pin but well, here we are. Maybe this is something to consider in #160 ?

martinberlin, 2022-02-27:
Exactly. I will add it to my list, thanks @vroland

martinberlin, 2022-06-02:
Reopening this since I rolled back proposed PR from Yardie
This merge in a single commit: fd8a290f68d360c33ba7833487f38566e23e7178
was reversed since it added a function **cfg_powerdown** is a exact copy of existing funcion **cfg_poweroff**

Since after reviewing it, I don't think is a good solution to the problem, and it should be done differently. Ultimately if this is a hardware issue from Lilygo, would be the best to make a different approach on it, after #183 is merged.

Yardie-, 2022-06-03:
Good idea @martinberlin. It will work cleaner and there will be documentation and examples when it does.

martinberlin, 2022-06-04:
Please try this as soon as #185 is resolved.

Yardie-, 2022-07-03:
Sorry @martinberlin  I have not checked this is a while. Basically this is now simply a documentation issue. There is a function that allows touch in the new code. I'll sort that and do a PR

closed 2022-07-03 by Yardie-