vroland/epdiy #164: epd_full_screen doesn't respect rotation

Yardie-, 2022-03-01:
Shouldn't it be this ( I left the original line in but commented out ) 
```
EpdRect epd_full_screen() {
  //~ EpdRect area = {.x = 0, .y = 0, .width = EPD_WIDTH, .height = EPD_HEIGHT};
  EpdRect area = {.x = 0, .y = 0, .width = epd_rotated_display_width(), .height = epd_rotated_display_height()};
  return area;
}

```

martinberlin, 2022-03-01:
Hello @Yardie- 
This is not an issue, since fullscreen is exactly the exception to calculate areas, when it's full screen it's the full area of the screen.
You can check this updating dragon example function to draw the image to:

    epd_draw_rotated_image(dragon_area, dragon_data, epd_hl_get_framebuffer(&hl));

And after the epd_init do a: epd_set_rotation(EPD_ROT_INVERTED_LANDSCAPE);

Then the Dragon should come inverted as expected. If I'm wrong post an example to reproduce this issue.

Yardie-, 2022-03-01:
Ok that's cool.
Just not what I was expecting.
I'll close this now. 
Thanks Martin (@martinberlin)

Yardie-, 2022-03-01:
Hi Martin @martinberlin 
Well having finally tried rotation I realise my error.
It certainly doesn't work as I expected. 

I built an app with text and hand drawn images. 
I expected to be able to just do 
epd_set_rotation(EPD_ROT_PORTRAIT);
and my whole app would respect the new orientation but nothing does.
I expected the code to work in that manner.

This is then clearly not the designed behavior so I will leave it there and work with what I have.
Thanks for your time and help.
Greg