vroland/epdiy #165: font.c addtion idea

Yardie-, 2022-03-01:
I have written a new function for font.c
add it if you like 
it's basically cut and paste with a few modifications.
needs to be added into 
epd_driver.h as well

```
EpdRect epd_get_string_rect (const EpdFont *font, const char *string,
                     int x, int y, const EpdFontProperties *properties )
{
  assert(properties != NULL);
  EpdFontProperties props = *properties;
  props.flags |= EPD_DRAW_BACKGROUND;
  EpdRect temp  = {.x = x, .y = y, .width = 0, .height = 0};
  if (*string == '\0') 
    return temp;
  int minx = 100000, miny = 100000, maxx = -1, maxy = -1;
  int temp_x = x;
  int temp_y = y;
  
  // Go through each line and get it's co-ordinates
  uint32_t c;
  while ((c = next_cp((const uint8_t **)&string)))
  {
    if(c==0x000A) // newline
    {
      temp_x = x;
      temp_y += font->advance_y;
    }
    else 
      get_char_bounds(font, c, &temp_x, &temp_y, &minx, &miny, &maxx, &maxy, &props);
  }
  temp.x = ( x < minx ) ? x : minx;
  temp.width = maxx - x;
  temp.y = miny;
  temp.height = maxy - miny;
  return temp;
}

```

martinberlin, 2022-03-01:
Hi Greg,
here you have a introduction into doing a pull request. 

**Configure remote from a fork**

The general practice is to add the original repo as Upstream:
https://help.github.com/articles/configuring-a-remote-for-a-fork/

As second step you can create in your Fork a branch that has same contents as this repository master, for example I call this branch vroland-master

```
git checkout -b vroland-master upstream/master
Branch 'vroland-master' set up to track remote branch 'master' from 'upstream'.
```

Then in this branch you propose this change in the code. And basically after pushing your changes you will get already in Github interface the chance to make a pull-request. Please add also why this change should be introduced and what is the enhancement it will bring. 

Creating a pull request

https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

Yardie-, 2022-03-01:
Thanks I was to lazy for so small amounts of code but I do the RTFM :)
Greg

Yardie-, 2022-03-02:
Thank you Martin (@martinberlin) I did actually spend the time to sort out my git sort of properly and add the pull request.