vroland/epdiy #170: ordering a board I got stuck in the layer order section.

Yardie-, 2022-03-15:
I love this and now want to build my own.
I came a little unstuck at the JLPCB page.
I uploaded the gerber zip and got to a section about the board layer order.
At that point I gave up.
Could someone please write a
 
"How to order a board for dummies please?"

I'm lost.

martinberlin, 2022-03-15:
Sent you an invitation to epdiy.slack.com so you can ask there in the boards section.
I already proposed time ago that this could be in the WiKi section.
But is not a bug or problem with the library so you can close this. 

I already told you some time ago that the Issues part of github is not a Frequent asked questions and that there are better channels for that. 
Of course will be really nice that this is documented but someone has to find the time to write it. Also in JLPCB there are not all the components, so you will have to buy those in Digikey and solder them yourself. But before ordering something complex like this you should get a bit of experience with JLPCB and the fabrication process, everyone get's stuck the first times.

Yardie-, 2022-03-15:
thank you @martinberlin for the invite as that is what is required for slack.
There is no clear way to find this information and I would sort of call that a bug.
But I will close this and get back in my box.

closed 2022-03-15 by Yardie-