vroland/epdiy #171: Greyscale mapping doesn't work as expected

aceisace, 2022-04-05:
Hi @vroland 
I've been testing the epdiy-library for quite some time, initially using the forked code of the 4.7" lilygo Display. When switching to this library, it seems there is a bug during rendering. The greyscales are not shown as expected. Some appear to be missing while others seem duplicated.

As a test and to help you with debugging, I first testet the greyscale test of lilygo:
This is how it should look like:
[image]
And this is how it is rendered with the lilygo library:
[image]

Hence it _should_ look the same with the epdiy-lib. But this is the result with the epdiy-lib:
[image]

To rule out that this issue was specific to the 4.7" display, I then tried it with the 9.7" one, with the following result:
[image]

Could you look into this please? Thanks in advance

Yardie-, 2022-04-22:
I'm a bit curious about this one. 
I have a couple of questions though. 
What was each screen before the image render. That is did they both have a clean white screen prior to drawing the image.
Did you try redrawing the image over itself.
I don't know a heap but sometimes strange things happen depending on the previous state of the screen.
Also did you try the different refresh types.
What I see here is a waveform issue rather than a code issue. Lillygo did GPL their waveform and it is in the epdiy release as it stands. So I'm curious as to what is going on here. Also maybe you could add the files necessary ie code and image so that others can look into this properly. Otherwise your request is pretty vague and hard to reproduce.
Just some ideas.

aceisace, 2022-04-22:
@Yardie- Hi and thanks for your comment.
Some additional info:
* All displays were cleared (white) before displaying these images
* Since the images are generated using maths, issues related to buffer-generation can be ruled out
* I have tested both waveforms, the ones provided by epdiy and original waveforms. Changing the waveform doesn't cause any changes. The lilygo lib works as expected. But epdiy doesn't seem to render it correctly
* Redrawing the image over itself does not cause any changes

Code used to render greyscales with the for 4.7" - requires the [lilygo-4.7" lib](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47)
```cpp
/* Simple firmware for a ESP32 displaying a static image on an EPaper Screen.
 *
 * Write an image into a header file using a 3...2...1...0 format per pixel,
 * for 4 bits color (16 colors - well, greys.) MSB first.  At 80 MHz, screen
 * clears execute in 1.075 seconds and images are drawn in 1.531 seconds.
 */

#include <Arduino.h>
#include "epd_driver.h"

uint8_t *grayscale_img;
uint8_t *grayscale_img2;

void setup()
{
    Serial.begin(115200);

    epd_init();

    // copy the image data to SRAM for faster display time
    grayscale_img = (uint8_t *)heap_caps_malloc(EPD_WIDTH * EPD_HEIGHT / 2, MALLOC_CAP_SPIRAM);
    if (grayscale_img == NULL) {
        Serial.println( "Could not allocate framebuffer in PSRAM!");
    }

    grayscale_img2 = (uint8_t *)heap_caps_malloc(EPD_WIDTH * EPD_HEIGHT / 2, MALLOC_CAP_SPIRAM);
    if (grayscale_img2 == NULL) {
        Serial.println( "Could not allocate framebuffer in PSRAM!");
    }

    uint8_t grayscale_line[EPD_WIDTH / 2];
    uint8_t value = 0;
    for (uint32_t i = 0; i < EPD_WIDTH / 2; i++) {
        uint8_t segment = i / (EPD_WIDTH / 16 / 2);
        grayscale_line[i] = (segment << 4) | segment;
    }
    for (uint32_t y = 0; y < EPD_HEIGHT; y++) {
        memcpy(grayscale_img + EPD_WIDTH / 2 * y, grayscale_line, EPD_WIDTH / 2);
    }

    value = 0;
    for (uint32_t i = 0; i < EPD_WIDTH / 2; i++) {
        uint8_t segment = (EPD_WIDTH / 2 - i - 1) / (EPD_WIDTH / 16 / 2);
        grayscale_line[i] = (segment << 4) | segment;
    }
    for (uint32_t y = 0; y < EPD_HEIGHT; y++) {
        memcpy(grayscale_img2 + EPD_WIDTH / 2 * y, grayscale_line, EPD_WIDTH / 2);
    }
}

void loop()
{
    uint32_t t1, t2;
    Serial.println("WHITE_ON_BLACK drawing:");
    epd_poweron();
    epd_clear();
    t1 = esp_timer_get_time();
    epd_push_pixels(epd_full_screen(), 20, 0);
    epd_push_pixels(epd_full_screen(), 20, 0);
    epd_push_pixels(epd_full_screen(), 20, 0);
    epd_draw_image(epd_full_screen(), grayscale_img2, WHITE_ON_BLACK);
[... 104 lines cut ...]
  }

  uint8_t grayscale_line[EPD_WIDTH / 2];
  uint8_t value = 0;
  for (uint32_t i = 0; i < EPD_WIDTH / 2; i++) {
    uint8_t segment = i / (EPD_WIDTH / 16 / 2);
    grayscale_line[i] = (segment << 4) | segment;
  }
  for (uint32_t y = 0; y < EPD_HEIGHT; y++) {
      memcpy(grayscale_img + EPD_WIDTH / 2 * y, grayscale_line, EPD_WIDTH / 2);
  }

  value = 0;
  for (uint32_t i = 0; i < EPD_WIDTH / 2; i++) {
    uint8_t segment = (EPD_WIDTH / 2 - i - 1) / (EPD_WIDTH / 16 / 2);
    grayscale_line[i] = (segment << 4) | segment;
  }
  for (uint32_t y = 0; y < EPD_HEIGHT; y++) {
      memcpy(grayscale_img2 + EPD_WIDTH / 2 * y, grayscale_line, EPD_WIDTH / 2);
  }

  xTaskCreate(&epd_task, "epd task", 10000, NULL, 2, NULL);
}

```

martinberlin, 2022-04-22:
Hello @aceisace this latest code you copied here will not compile with latest epdiy version.
As you can see in the codebase epd_init function requires a parameter so trying to build this you will get:

too few arguments to function 'void epd_init(EpdInitOptions)'
   epd_init();
   
 Can you please post a version with the latest stand of epdiy where we can reproduce this?

If you try to compile the grayscale test of this repository
https://github.com/vroland/epdiy/tree/master/examples/grayscale_test

The demo works rendering the grayscale correctly with both Waveforms. I will make an Pull request soon adding Lilygo old Waveform as a default since it is faster and leaving the alternative new version as optional.

mcer12, 2022-04-22:
@aceisace  This seems exactly like my issues with vendor waveforms. Please try to replace the 4.7" display waveform with the one from lilygo fork. I bet it will work!

aceisace, 2022-04-23:
@martinberlin Thanks for your comment!
Yes, you are right, the first code is directly form the lilygo fork and will only work with the current lilygo-fork, but not with epdiy. To test the first code, you'll have to remove the epdiy-lib from the Arduino IDE libraries folder, add the lilygo-fork for the 4.7" Display and then run the code. But that exactly _is_ the problem, I'll explain:

I've bought the lilygo 4.7" display some time ago. I ran some tests to check if all 16 greyscales could be rendered (see first code of my previous comment). Everything worked as expected. Eventually I wanted to test a 9.7" display, but the lilygo fork doesn't support the 9.7" display, so it was time to switch from the lilygo-fork to the epdiy repo. 

Thanks to the boards from @mcer12 (awesome boards btw!), I managed to display some images on the 9.7" display using epdiy. But the rendering on the 4.7" and 9.7" didn't look half as good as I when was using the lilygo-fork. This was without modifications.

What I had expected is that the epdiy repo would render all 16 greyscales just like the lilygo-fork, but it didn't (see images in first comment)

I suspected the built-in waveforms of epdiy _maybe_ didn't support displaying all 16 greyscales, so I used inkwave to generate headers from original waveforms. I can confirm the new waveforms were added to epdiy successfully and that they are actually being used when I select the 9.7" board and during rendering (due to a different flashing pattern while updating). But even then, the 9.7" display did not show all 16 greyscales when running the following demo code:
```cpp
/* Inkster arduino example -  */

#include "driver/adc.h"
#include "esp_adc_cal.h"

#include <Arduino.h>
#include <NeoPixelBus.h>

extern "C"
{
#include "epd_driver.h"
#include "epd_highlevel.h"
}

uint8_t *framebuffer;
uint64_t wakeUpPins = GPIO_SEL_13 | GPIO_SEL_36 | GPIO_SEL_39; // Wakeup pins, same as button pins
uint8_t *grayscale_img;

EpdiyHighlevelState hl;
uint8_t *fb; // EPD 2bpp buffer

int deepSleepSeconds = 600;
int temperature = 25;

void checkWakeupReason()
{
    esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();

    if (wakeup_reason == ESP_SLEEP_WAKEUP_EXT1)
    {
        uint64_t rst_pin = esp_sleep_get_ext1_wakeup_status();

        if (rst_pin == GPIO_SEL_36)
        {
            Serial.println("[SYSTEM] Wakeup by button 1");
        }
        if (rst_pin == GPIO_SEL_39)
        {
            Serial.println("[SYSTEM] Wakeup by button 2");
        }
        if (rst_pin == GPIO_SEL_13)
        {
            Serial.println("[SYSTEM] Wakeup by button 3");
        }
    }
    else if (wakeup_reason == ESP_SLEEP_WAKEUP_TIMER)
    {
        Serial.println("[SYSTEM] Wakeup by timer");
    }
}

void deepSleep()
{
    epd_poweroff();
    epd_deinit(); // needs to be called before sleep for low power consumption

    Serial.print("[SYSTEM] Going to sleep for ");
    Serial.println(deepSleepSeconds);
    Serial.println(" seconds.");

    esp_sleep_enable_ext1_wakeup(wakeUpPins, ESP_EXT1_WAKEUP_ANY_HIGH);

    esp_sleep_enable_timer_wakeup(1000000LL * deepSleepSeconds);

    esp_deep_sleep_start();
}

void loop()
{
    /* wow so much empty */
}

void setup()
{

    Serial.begin(115200);
    Serial.println();

    // copy the image data to SRAM for faster display time
    grayscale_img = (uint8_t *)heap_caps_malloc(EPD_WIDTH * EPD_HEIGHT / 2, MALLOC_CAP_SPIRAM);
    if (grayscale_img == NULL) {
        Serial.println( "Could not allocate framebuffer in PSRAM!");
    }

    checkWakeupReason();

    epd_init(EPD_LUT_64K);
    hl = epd_hl_init(EPD_BUILTIN_WAVEFORM);
    fb = epd_hl_get_framebuffer(&hl);

    epd_poweron();

    unsigned long updateMillis = millis();

    // If you want to make multiple screen updates, run epd_fullclear before epd_copy_to_framebuffer

    Serial.print("Clearing Display...");
    epd_fullclear(&hl, temperature);
    // epd_hl_update_screen(&hl, MODE_GC16, temperature);
    Serial.println("Done");

    
    uint8_t grayscale_line[EPD_WIDTH / 2];
    uint8_t value = 0;
    for (uint32_t i = 0; i < EPD_WIDTH / 2; i++) {
        uint8_t segment = i / (EPD_WIDTH / 16 / 2);
        grayscale_line[i] = (segment << 4) | segment;
    }
    for (uint32_t y = 0; y < EPD_HEIGHT; y++) {
        memcpy(grayscale_img + EPD_WIDTH / 2 * y, grayscale_line, EPD_WIDTH / 2);
    }

    Serial.print("Displaying image..");
    epd_copy_to_framebuffer(epd_full_screen(), grayscale_img, epd_hl_get_framebuffer(&hl));
    epd_hl_update_screen(&hl, MODE_GC16, temperature);
    Serial.println("Done");

    Serial.print("[SYSTEM] Display update took ");
    Serial.print((millis() - updateMillis));
    Serial.println("ms");

    epd_poweroff();
    //epd_deinit();

    deepSleep();
}
```

What happens when you run the above code on your end?  Does it show all 16 greyscales on your displays? Thanks in advance

martinberlin, 2022-04-23:
> To test the first code, you'll have to remove the epdiy-lib from the Arduino IDE libraries folder, add the lilygo-fork for the 4.7" Display and then run the code

> Code used to test the 4.7" with epdiy:

No I meant to test the second part, where you claim that is EPDiy, but it’s not possible to build it with current EPDiy.
As per rendering images with 16 grayscales I didn’t had any problems with the current library, here is an image rendered from a bitmap, on a board that had one data line dead (hence one live every 8 is white)
[image]

aceisace, 2022-04-23:
Thanks, looking forward to the rendered result on your end!
The rendered image you posted looks pretty good. Just a question regarding the bottom left area of the image, that part looks a bit blurred, does the original image look the same too? Often it's hard to tell if the rendered image really does use all 16 greyscales, hence the test with the greyscales.

Thanks for the link to your bmp-reader, certainly helps with getting started in parallel epds :). I'm just using a modified version of [imgconvert](https://github.com/vroland/epdiy/blob/master/scripts/imgconvert.py) to generate the required buffers

martinberlin, 2022-04-23:
Just adding some Serial output to the for loop I get this gray values and image looks similar to the second one you posted. 
This is testing with the new waveform of Lilygo that is not what they use in their repository (The old EPDiy version they use does not support waveform I think)

```
Clearing` Display...Done
i 0 gray: 0
i 30 gray: 17
i 60 gray: 34
i 90 gray: 51
i 120 gray: 68
i 150 gray: 85  --> Here there is a step jump
i 180 gray: 102
i 210 gray: 119
i 240 gray: 136
i 270 gray: 153
i 300 gray: 170
i 330 gray: 187
i 360 gray: 204  --> This is the gray that looks darker
i 390 gray: 221
i 420 gray: 238
i 450 gray: 255
```
Now when I use the grayscale demo that is in this repository this is what I get:
https://github.com/vroland/epdiy/blob/master/examples/grayscale_test/main/main.c

In this image, the 4th gray is also a bit off like you can see in the image, but not so much like in the example you posted. Probably there is something wrong with the gray timings, so someone will have to take the time to correct the waveform.

martinberlin, 2022-04-23:
Please try using this repository branch:

https://github.com/martinberlin/epdiy-rotation/tree/vroland-master

And render the grayscales again, here is my result now. Pull-request #172

Yardie-, 2022-04-24:
@aceisace Great example code.

aceisace, 2022-04-24:
@Yardie- Thanks, it's just a modified version of the demo code for the inkster board from @mcer12 :)

aceisace, 2022-04-24:
[image]
I guess this counts as working, @martinberlin :)

I used my own demo code from the previous comment since the other one isn't an .ino file and can't be used directly with Arduino IDE.

My 4.7" epd looks a bit pale in comparison to yours. Probably dependent on VCOM. Are there any tips to get VCOM calibration as precise as possible? I know there was a code somewhere, but I think it wasn't compatible with Arduino IDE.

In any case, I'll test the greyscale code on the 9.7" too since it had pretty similar issues (see images from first comment)

aceisace, 2022-04-24:
Okay, seems like the 9.7" is working too now:
[image]
I'll see what happens when I replace the waveforms for the 9.7" with original vendor waveforms

closed 2022-04-30 by martinberlin