vroland/epdiy #174: Will LGPL license be changed to MIT?

lanistor, 2022-04-29:
Thank @martinberlin for developed (driver of lvgl)[https://github.com/martinberlin/lvgl_epaper_drivers], connected epdiy with the nearly best embeded GUI library. 
But the code cannot be merged because of LGPL license of epdiy: https://github.com/lvgl/lvgl_esp32_drivers/pull/75#issuecomment-1112135305

So will the license be changed to MIT? I know the license had changed from  GPL, but this still doesn't look friendly enough for open source contributions.

martinberlin, 2022-05-05:
Hi Lanistor, in #138 @vroland mentioned about the License being GPL:

> The main reason is that I wanted derivative works to stay open-source.

I guess would be his ultimate decision what kind of License to use for his repository.

martinberlin, 2022-05-06:
By the way I would like that my proposal is merged one day and even take the time to explore how to make it work in future LVGL versions.
But all this is pointless if it will never be merged because there is a License conflict, something that is really not under my control as a developer. Everyone has his reasons to choose the right License for a project and a decision we should respect. Let's wait some weeks to know what are @vroland thoughts on this topic.

lanistor, 2022-05-07:
Let's wait patiently for @vroland 's reply

lanistor, 2022-08-18:
@vroland Hi, what do you think of changing license of `epdiy` to MIT ?

vroland, 2022-09-02:
For now I'm still leading towards LGPL. IMO, it is permissive enough to allow linking to other code bases, which I see is a requirement for many people. For epdiy itself however, I was glad to have copyleft when lilygo forked the driver. That way, they had to publish all changes and we could make epdiy compatible. 
With LVGL, I don't see a problem with LGPL, even with static linkage: https://www.gnu.org/licenses/gpl-faq.html#LGPLStaticVsDynamic
Since they provide the source code, it is easy to re-link different another version of epdiy or vice-versa. So I don't see any conflict here, even with static linking.

lanistor, 2022-09-03:
@vroland I understand the original intention of your choice of LGPL, and I also agree with it.

However, do you think some developers who use epdiy for commercial products without fork epdiy, what should they do better? They used epdiy to realize the rendering of epaper, but they designed many other functions by themselves to realize a complete commercial product. In order to protect their product, they need to apply for a patent.

They will face a problem, they must dynamically link epdiy to protect their products, otherwise others can copy their products. So they must to find another ways to avoid this problem.

In fact, we are facing such a problem, and i think this is also the reason that LVGL don't allow https://github.com/lvgl/lvgl_esp32_drivers/pull/75#issuecomment-1112135305 to merge.

martinberlin, 2023-08-28:
Thanks.
Will try to see if I can achieve somehow adding epdiy as a component with static linkage as you describe, 
Lanistor I’m closing here since your question was answered.

closed 2023-08-28 by martinberlin

lanistor, 2023-08-29:
Got it, thanks a lot.