vroland/epdiy #185: Compile errors with Arduino

Yardie-, 2022-06-04:
Trying to compile throws
```
Arduino/libraries/epdiy/src/epd_driver/board/epd_board_v6.c:2:10: fatal error: board/epd_board_v6.h: No such file or directoryResolveLibrary(board/epd_board_v6.h)
  -> candidates: []
 #include "board/epd_board_v6.h"

```
I know the Arduino precompiler is bizarre and I have often had path issues like this. 
It is something I never really solved.

martinberlin, 2022-06-04:
As I commented in Slack I think the issue with Arduino IDE is that it expects all header includes  without relative paths.
Gregory also please add exact steps to reproduce the error. Like: 

- 1 Setup https://github.com/vroland/epdiy.git as a library ie symlink epdiy git to ~/Arduino/libraries
- 2 Create an Arduino basic sketch to start display, draw some text or shapes and clearscreen at the end

mickeprag, 2022-06-04:
Ok, so I have been able to reproduce the issue. I get the same error.
I wonder what the best solution for this is? Is is possible to specify include directories for Arduino libraries?

mickeprag, 2022-06-04:
Adding the include paths explicitly is unfortunately not yet supported:
https://github.com/arduino/arduino-cli/pull/502

PR #187 (2022-06-05): Include using full relative path
This is needed by Arduino since it is not yet supported to specify
include directories manually

Fixes #185
closed 2022-06-05 by martinberlin