vroland/epdiy #186: Possibility of integration into the smart home?

19leva89, 2022-06-04:
Good afternoon, tell me if it is possible to integrate this device into smart home control systems, for example Home Assistant? Or data transfer via MQTT protocol?

mickeprag, 2022-06-04:
This is just a library for integration into other softwares. You need to write the software to integrate with HA or MQTT yourselves.

mickeprag, 2022-06-04:
To clarify.
Is it possible? Yes it is!
Is there a solution ready to be used as-is? No, not that I am aware of.

martinberlin, 2022-06-04:
Please check the https://epdiy.slack.com channel. There you can find @ashald that I think he used LilygoEPD47.
Closing here since this library is just the Epaper controller and per se has nothing to do with home assistance. You are welcome to join the slack group if the invitation link is inactive please send me your email and will invite you ( martin@luckycloud.de )

closed 2022-06-04 by martinberlin

ashald, 2022-06-04:
@19leva89 see https://github.com/esphome/feature-requests/issues/1109 - there is a custom component for ESPHome, that allows you to use Lilygo T5-4.7 with HomeAssistant. Or ping me on Slack with any questions you might have.