vroland/epdiy #188: Lilygo T5 + ESPHome - Error

xcybplx, 2022-06-14:
Hello,
Some time ago I created an esphome yaml file using vrolad component and it was working fine. No code on my side was changed but now when I'm trying update the code I have some errors. Can anyone help with that? I don't know what I can do with it:

Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/board/epd_board_v6.c.o
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c: In function 'epd_board_init':
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c:102:7: error: 'i2c_config_t {aka struct <anonymous>}' has no member named 'clk_flags'
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
       ^
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: error: 'I2C_SCLK_SRC_FLAG_FOR_NOMAL' undeclared (first use in this function)
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
                    ^
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: note: each undeclared identifier is reported only once for each function it appears in
Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/render.c.o
Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/rmt_pulse.c.o
*** [/data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/board/epd_board_v6.c.o] Error 1

martinberlin, 2022-06-14:
At simple view this is using the wrong board (V6)
Can you double check what is your board configuration?
Maybe copy below your platformio.ini file 👇

mickeprag, 2022-06-14:
@martinberlin Since there is no defines anymore all boards will be built regardless of what board you have defined....

xcybplx, 2022-06-14:
I would love to help you, but I'm using Esphome addon for homeassistant, it's not like arduino or something. 
I'm using simple yaml file, which only contains things like:

external_components:
  - source: github://ashald/esphome@lilygo-t5-47
    components:
      - lilygo_t5_47

And the rest is done automatically by esphome.

The full log below:

INFO Detected timezone 'Europe/Warsaw'
INFO Generating C++ source...
INFO Compiling app...
Processing lilygo-t52 (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
|-- <epdiy> 1.0.0+sha.4ad5615
Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/src/main.cpp.o
Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/board/epd_board_v6.c.o
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c: In function 'epd_board_init':
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c:102:7: error: 'i2c_config_t {aka struct <anonymous>}' has no member named 'clk_flags'
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
       ^
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: error: 'I2C_SCLK_SRC_FLAG_FOR_NOMAL' undeclared (first use in this function)
   conf.clk_flags = I2C_SCLK_SRC_FLAG_FOR_NOMAL;
                    ^
/data/lilygo-t52/.piolibdeps/lilygo-t52/epdiy/src/epd_driver/board/epd_board_v6.c:102:20: note: each undeclared identifier is reported only once for each function it appears in
*** [/data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/board/epd_board_v6.c.o] Error 1
Compiling /data/lilygo-t52/.pioenvs/lilygo-t52/libcb8/epdiy/tps65185.c.o
Archiving /data/lilygo-t52/.pioenvs/lilygo-t52/libFrameworkArduinoVariant.a
========================= [FAILED] Took 22.87 seconds =========================

xcybplx, 2022-06-14:
It may be related to this change: https://github.com/espressif/arduino-esp32/releases
Latest adruino-esp32 is using new ESP-IDF

martinberlin, 2022-06-14:
I see Micke thanks for the reminder.
Try to comment that line. And build again
I think I hit this once. But what version of IDF is?
It should be a way to know this...no matter what you are using to compile it.

xcybplx, 2022-06-14:
Based on the informations from the website: ESP32 Arduino 2.0.3 based on ESP-IDF 4.4.1

martinberlin, 2022-06-14:
Anyways just for your information: 
$ idf.py --version
ESP-IDF v4.4.1

$ grep -rne "I2C_SCLK_SRC_FLAG_FOR_NOMAL" .

This should be in your version of IDF if it's 4.4 (Using this one also)
./components/driver/include/driver/i2c.h:35:

#define I2C_SCLK_SRC_FLAG_FOR_NOMAL       (0)         /*!< Any one clock source that is available for the specified frequency may be choosen*/

So just change that I2C_SCLK_SRC_FLAG_FOR_NOMAL for a 0 and try to build again. No idea why you get this message if you are really using this version-

mickeprag, 2022-06-14:
If the actual error is `I2C_SCLK_SRC_FLAG_FOR_NOMAL` that is missing then we might need to add an extra include in our board file.

Anyway, try changing the line to 0 as @martinberlin suggest then we will know.

xcybplx, 2022-06-15:
Unfortunatelly, I don't know how to do this on esphome thing ;/

martinberlin, 2022-06-15:
There must be a way to control what is going on with the components versioning in esphome. Because if there is not this is more a problem from esphome than this library.
@ashald do you have some idea on how to resolve this?

xcybplx, 2022-06-15:
I made a mistake, component from @ashald is working fine, this one give those errors:

external_components:
  - source:
      type: git
      url: https://github.com/vbaksa/esphome
      ref: dev

martinberlin, 2022-06-15:
It was not backward incompatible, but things need to evolve. Please adjust the esphome part or use the version 1.0.1 that was tagged just before this refactoring:
https://github.com/vroland/epdiy/releases/tag/1.0.1
Unless there is something else to discuss we should close here. I cannot reproduce this error compiling with the esp-idf 4.4 neither with other firmwares that use this as platformio library (using latest arduino-esp32)
Will do some more tests tomorrow but unless more details are provided, there is no feasible way to solve this. I do not have to install esphome to find this, please give us a way to know exactly what is going on and to reproduce it.

xcybplx, 2022-06-16:
@ashald I was able to fix this using your tips but ESPHome was updated, and now I have issues even with your component...

```
Compiling /data/lilygo-t5/.pioenvs/lilygo-t5/src/main.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:1973:44: error: invalid new-expression of abstract class type 'esphome::lilygo_t5_47::LilygoT547Display'
   t5 = new lilygo_t5_47::LilygoT547Display();
                                            ^
In file included from src/esphome.h:31:0,
                 from src/main.cpp:3:
src/esphome/components/lilygo_t5_47/display/lilygo_t5_47_display.h:13:7: note:   because the following virtual functions are pure within 'esphome::lilygo_t5_47::LilygoT547Display':
 class LilygoT547Display : public PollingComponent, public display::DisplayBuffer {
       ^
In file included from src/esphome.h:21:0,
                 from src/main.cpp:3:
src/esphome/components/display/display_buffer.h:373:23: note: 	virtual esphome::display::DisplayType esphome::display::DisplayBuffer::get_display_type()
   virtual DisplayType get_display_type() = 0;
                       ^
*** [/data/lilygo-t5/.pioenvs/lilygo-t5/src/main.cpp.o] Error 1
```

martinberlin, 2022-06-21:
This is not strictly related to EPDiy component and something it has to be solved on the ESPHome side. If it is and I made a mistake here please re-open this with proper hints in how to fix this in the component side.