vroland/epdiy #189: Source file comment issue in rmt_pulse.c

Li-Tianming, 2022-06-15:
The comment said `// currently hardcoded: use channel 0 `

https://github.com/vroland/epdiy/blob/4ad561586916093a4b8bd0a7ee94485a3ac23cd9/src/epd_driver/rmt_pulse.c#L24

but actually it use `CHANNEL_1`

https://github.com/vroland/epdiy/blob/4ad561586916093a4b8bd0a7ee94485a3ac23cd9/src/epd_driver/rmt_pulse.c#L25

i tried other channle like`CHANNEL_4` , it workes.

so maybe it's just a misspelling?

martinberlin, 2022-06-15:
Hi Li,
Please provide clear information on how to reproduce this.
What board you are using?
With what frnmework? (Arduino-esp32/esp-idf)

Li-Tianming, 2022-06-16:
I use V5 board and esp-idf V4.4

I think comment in this line 

https://github.com/vroland/epdiy/blob/4ad561586916093a4b8bd0a7ee94485a3ac23cd9/src/epd_driver/rmt_pulse.c#L24

is wrong

it should be 

` // currently hardcoded: use channel 1`

closed 2022-06-21 by martinberlin