vroland/epdiy #190: Really hope to support ES133TT3！

geekheart, 2022-06-28:
I saw this [news ](https://news.ycombinator.com/item?id=2624556)and became interested in the ink screen and bought ES133TT3. But there is no information, and Google searched for a long time and couldn't get any relevant information.

martinberlin, 2022-06-29:
Hi @geekheart the link you are providing is going to a Hacker news site that has nothing to do with the epaper specs.
Please provide a link to the datasheet of this epaper so we can take a look.

geekheart, 2022-06-29:
[ES133TT3_Ver2.0_20161011_201801119664.pdf](https://github.com/vroland/epdiy/files/9011648/ES133TT3_Ver2.0_20161011_201801119664.pdf)
Here is the only information I found

martinberlin, 2022-06-29:
[image]
Sadly this won’t work since it has 16 data lines. There is simply not enough gpios in ESP32-WRover for this one, if someone would make a special version with S2 or S3 it would work, but it will take months or even years to get there.
I suggest you to check a controller that suits the display you have and to get another compatible epaper if you want to test EPDiy.

closed 2022-07-07 by martinberlin