vroland/epdiy #195: IDF RGB Lcd controller component?

errolt, 2022-07-11:
Sorry if this is the wrong place to ask, but would the RGB LCD controller not simplify the code a lot? It already allows for sync front porch/back porch etc. The only I can see this far is that it is meant to  be set up and left running, where as the EPD is updated in individual frames?

Example: https://github.com/espressif/esp-idf/blob/9b8c558e63d95b491355a22e69f247834e899b47/examples/peripherals/lcd/rgb_panel/main/rgb_lcd_example_main.c

I'm still reading through the code to try and figure out how everything works, in an attempt to port it to ESP32-S3, so I might be misunderstanding things...

errolt, 2022-07-20:
Sorry, I'm an idiot. Looks like this is only supported on the S3

closed 2022-07-20 by errolt

martinberlin, 2022-07-21:
No it's actually the way to go @errolt, if there is any evolution on this component in future hardware revisions, the way to go will be to use S3.