vroland/epdiy #196: RTCIO error: rtc_gpio_isolate: RTCIO number error

hpsaturn, 2022-07-12:
Hi everyone, first to say that it's a awesome development. Congrats!

## Overview

I have the next execution message error every time that the panel is refreshed, it isn't critical but is weird. I think that it appeared in the last versions of Espressif framework.

## Output

```cpp
 E (4322) RTCIO: rtc_gpio_isolate(171): RTCIO number error
```

## Config setup

```python
platform = espressif32
board = esp32dev
framework = arduino
```

## Dependencias and packages

```python
Resolving epd47esp32 dependencies...
Platform espressif32 @ 5.0.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20003.220626 (required: platformio/framework-arduinoespressif32 @ ~3.20003.0)
├── tool-cmake @ 3.16.4 (required: platformio/tool-cmake @ ~3.16.0)
├── tool-esptoolpy @ 1.30300.0 (required: platformio/tool-esptoolpy @ ~1.30300.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
├── tool-openocd-esp32 @ 2.1100.20220411 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── toolchain-esp32ulp @ 1.22851.191205 (required: platformio/toolchain-esp32ulp @ ~1.22851.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3)

Libraries
├── ArduinoJson @ 6.19.4 (required: bblanchon/ArduinoJson @ ^6)
├── NTPClient @ 3.1.0+sha.aafa4e1 (required: git+https://github.com/taranais/NTPClient.git)
├── epdiy @ 1.0.0+sha.b4c582b (required: git+https://github.com/vroland/epdiy.git)
└── esp32FOTA @ 0.1.4+sha.d1bbc3f (required: git+https://github.com/hpsaturn/esp32FOTA.git)
``` 

# Hardware

Chip is ESP32-D0WD-V3 (revision 3)  
[LilyGo-EPD47](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47)

martinberlin, 2022-07-17:
@hpsaturn please checkout my branch in this PR and build that in the Lilygo so you can confirm that works

https://github.com/vroland/epdiy/pull/197

This will be merged only if someone else tests it.

hpsaturn, 2022-07-17:
Yes, working like a charm. Thanks.

martinberlin, 2022-07-17:
Reopening till I merge the PR tomorrow so you can go back,to use the official repository.