vroland/epdiy #199: Every X pixels in a row not being drawn to screen

dot4qu, 2022-07-25:
I'm seeing an annoying issue where no matter what I draw to the screen, there are white lines every X column where X is a small, consistent number of pixels. I think a picture will describe what I'm talking about better than words:

[image]

I'm 99% sure this is **not** an issue with epdiy and is a hardware problem with my board. Most likely a single bad solder joint on the FPC connector or perhaps just enough capacitance on a trace to the display that causes a signal to be a bit late in arriving. 

I've seen this behavior with multiple boards I've assembled and different displays. I know they *can* work properly as I've seen totally normal operation too. Interestingly enough, one of my boards worked perfectly fine for the first few days, then devolved into this behavior and hasn't changed since. Which is why I'm very confident it's a HW issue on my end and not the library.

My reason for creating an issue is just to ask what you might think the issue would be. I'm not familiar enough with the protocol for all the different signals that clock in a row of data, write to the display buffer, advance to the next row, etc. Since you're much more familiar with the algorithm involved in writing data, perhaps you know off the top of your head what a likely cause is for this?

It's my understanding that data is written row by row, not column by column. So somehow every X bits written from the frame buffer to the display are getting dropped.

mickeprag, 2022-07-25:
I agree this looks like a hardware issue.

The display is rendered row by row and column by column, just as you say. But each column is 8 pixels and each pixel have its own data signal. So it looks like one or several of the data (d0..d1) signals is broken. Either in the display, cable or board.

dot4qu, 2022-07-25:
ahhh okay that's exactly the info I was looking for, I can throw a scope on d0-d7 to see if any are acting up. Thanks!

dot4qu, 2022-07-25:
Closing this for now and will report back if I find any results for future searches

closed 2022-07-25 by dot4qu

dot4qu, 2022-07-26:
Sure enough, taking a look at the stretch of castellated edges where my epdiy data pins came out of the esp32, they looked a bit low on solder from reflow. Dumped a bunch in there by hand and voila, no more vertical column artifacts! Extremely happy it was an issue on the esp32 side and not the fine pitch flex connector.