vroland/epdiy #202: Will this support the ES108FC1 screen?

richild999, 2022-08-28:
es108fc1 
[image]

martinberlin, 2022-08-28:
If I see this correctly:

Interface Type | Parallel Data (2ch,  8-bit)                                                                                             , 50 pins                        FPC

This expects 2 ch of 8 bit. So it has 16 data lines. With the actual EDPiy V6 this is not supported.
Only a future ESP32S3 version with support of 16 data lines will provided you can supply a Waveform.

vroland, 2022-09-02:
This display looks pretty nice, so I've ordered one for testing once we work on the 2-channel version :) Note that this will probably take a while, though.

tomek-szczesny, 2022-10-11:
Does anyone have a datasheet of this thing?

mtreding, 2023-01-01:
How is this coming? Just bought one of these screens and would love to be able to use it for a project I’m working on for a static display. Thanks!

mtreding, 2023-01-01:
> This display looks pretty nice, so I've ordered one for testing once we work on the 2-channel version :) Note that this will probably take a while, though.

How’s it going? I just bought one of these, want to incorporate into an ongoing project.

vroland, 2023-01-01:
I have one too and plan to order an adapter board with the v7 prototype once it is done. Hope that's soon, but we'll see ;)

vdp, 2023-01-04:
Oh, look [what](https://github.com/vroland/epdiy/files/10342125/ES108FC1_Ver0.1.pdf) Santa brought me!

I only glanced through the DS. Things I noticed:

* if the max source driver frequency(60MHz in this case) is any indication the screen should be reasonably responsive

* not sure I like the recommended connector(Panasonic AXE550127D)- not the "clamshell" type

* requires 28V gate voltage, instead of the usual 22V. TPS65185(1) apparently supports this(table on pg 10 in their DS), by using a different value for one of the resistors in the voltage divider on the VDDH_FB pin, if I'm not mistaken. @vroland  perhaps it's worth considering making the resistor in question reasonably tweakable in the v7? At the very least I'd make the pads bigger, say at least 0805.

By the way has anyone already received their screen? How is the contrast(if there is some sort of image left on it)? 

Anyone tried if they can actually drive the screen? Haven't thought much about this but, in anticipation of v7, you can probably try it with the EPDiy you already have by preparing a custom FPC adapter, where the destination data lines are duplicated in respect to the source lines, so that every pixel is "cloned" in horizontal direction. The result probably won't be pretty nor practical, but at least you'd be able to test if it's working at all(after changing the aforementioned R, of course)

tomek-szczesny, 2023-01-04:
Frankly I'm lurking this thread because I plan to use this display with RK3568 SoC, that has built-in E-ink driver.
I hope to exchange information about the communication protocol of this particular model. Seems like nobody else on the planet is interested in this.

richild999, 2023-01-05:
I have the specification and I'm glad to share that. About the other information I'm very sorry that I have no more.

vdp, 2023-01-05:
@tomek-szczesny this actually sounds interesting. Are there any ready-made boards with this chip and suitable EPD connectors/power, that one can buy?

alexose, 2023-02-10:
Lurking this thread as well.  As far as I can tell, ES108FC1 has the best price/performance of any e-ink screen.  Would love to use two of them to build a foldable e-newspaper :)

vdp, 2023-02-20:
Hi, can someone that already received their screen confirm the type of its connector. The datasheet says that the panel's connector is AXE550127D(female/socket), which means for the board I have to order AXE650124D(male/header). Is that correct? 

Thanks!

richild999, 2023-02-21:
> Hi, can someone that already received their screen confirm the type of its connector. The datasheet says that the panel's connector is AXE550127D(female/socket), which means for the board I have to order AXE650124D(male/header). Is that correct? 
> 
> Thanks!
> 

[image]

[image]

[image]

[image]

richild999, 2023-02-21:
datasheet
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

richild999, 2023-02-21:
The full datasheet youcan see the repo : [ES108FC1](https://github.com/richild999/ES108FC1)

vdp, 2023-02-21:
Thanks a lot! I think the screen has the female/socket connector(AXE550127D), so the DS is correct. Too bad this may delay the release date for the v7 board, because I believe @vroland mistook it for a male/header(they are indeed hard to tell apart).

vdp, 2023-04-24:
Has anyone succeeded to drive this screen? I'm tinkering with my custom controller at the moment, and while I'm yet to test thoroughly the code, the results so far make no sense at all...

richild999, 2023-04-30:
> After looking at the waveforms using a logic analyzer, I see nothing wrong with them. Either the screen I've got is defective or, say, the pinout for ES108FC1 is different from the one given in the (preliminary) datasheet. Or perhaps there is a reason why these screens are so (relatively speaking) cheap - e.g. because they are junk, plain and simple... 
> 
> @vroland , have you been successful with your ES108FC1? 
> 

He has already successful on the screen
Link:
https://github.com/Hanley-Yao/Zynq7010_eink_controller
【板子刚刚修好浅试ED097TC2准备点ES108FC1-哔哩哔哩】 https://b23.tv/av772024193/p1

vdp, 2023-04-30:
Thank you! It didn't occur to me to check that project, although I've seen it before. I find the repo a bit hard to navigate unfortunately. I think he just dumped his Vivado(presumably- not familiar with Xilinx's tools) project folder to GH, so when I looked previously it wasn't obvious to me how it all fits together. I guess, I'll have to look closer and try to identify the relevant Verilog source files. I've spent more time than I planned, trying to make the ES108FC1 work, and moved on to other things now, but perhaps I should revisit it soon. And perhaps, I should try to contact the author.

You seem to understand Chinese- could please tell me what does he say in this video:

https://www.bilibili.com/video/BV1Ud4y1o7JD/

I saw similar vertical stripes pattern when testing my screen, though in my case it was updating only at seemingly random times and not as consistently, on every button press as his. That made me doubt if the gate driver works properly, because at least on my screen all gate driver MOSFETs(or whatever) are apparently turned on at the same instant.

richild999, 2023-04-30:
> Thank you! It didn't occur to me to check that project, although I've seen it before. I find the repo a bit hard to navigate unfortunately. I think he just dumped his Vivado(presumably- not familiar [...]

In that video, he says don't know what happend,the screen can't show the right view.

vdp, 2023-05-01:
Thanks!
I think I know what his initial problem was. He probably neglected to add a pull up resistor to the test(XON) pin as noted in the DS. I don't understand what the subtitles say, but I think in the beginning of the following video he shows how it was fixed:

https://www.bilibili.com/video/BV1qU4y1r7uo/

Unfortunately, this is not the problem I have. My screen just behaves erratically, so I think it may be broken.

tomek-szczesny, 2023-05-01:
@vdp, I'm far behind any of you in my e-ink endeavor, still got another project to finish before this, but as a pro EE I'll tell you this: People are always too quick to blame the hardware. Out of all things that could go wrong with your setup, this is not the most likely reason why it doesn't work.

The movie that you linked could be the answer. Maybe one signal needs pull-up despite it's not documented. Maybe he compensated for poor layout. Who knows? A Mandarin speaker, I guess. :)
I translated a few of these subtitles and the author did not explain what he did, just said it helped.

How about we set up a space for taking about this e-ink display instead of pestering someone's github issues. :) You guys wanna set up a Discord channel, or a hackaday chat?

martinberlin, 2023-05-01:
There is already EPDiy.slack.com

vdp, 2023-05-02:
Thanks for your reply and suggestions @tomek-szczesny !

> People are always too quick to blame the hardware. Out of all things that could go wrong with your setup, this is not the most likely reason why it doesn't work.

This is likely true in general for electronic goods that pass through reasonable QA process, but I'm not sure how well it applies to eink panels sold on AliExpress, unfortunately. As far as I know many of these screens are refurbished, produced by "ghost shifts" etc. Moreover, I don't know how your screen came, but mine was just packaged in bubble wrap, so nothing like metallized anti-ESD bag. Of course, this doesn't mean that it's impossible that I made mistake somewhere, though I tried to check that the waveforms look reasonable, and that there are no shorts or loose solder joints on the PCB and the panel adapter... I'll probably check once again at some point in near future. I may also try to contact @Hanley-Yao, and to ask him about his experience with this screen.

As for the idea for a forum, I agree with Martin that the existing Slack may serve the purpose, though I don't like that you can't see messages that are more than couple of months old without paying them a subscription fee. For me this is a __very__ serious drawback, as there are past discussions I'd like to review, and there is no hassle-free way to do that.

alexose, 2023-08-30:
Just to bump this thread... is this project still targeting ES108FC1 support?  If so, should it be added to the table in the README.md?

vroland, 2023-08-31:
I realized that the connector I wanted to order for this display didn't come with my parts order because it is out of stock. And I don't really see it elsewhere :( So I cant make an adapter board to try. Do you have any source for the connector?

martinberlin, 2023-11-06:
I‘ll try to make an adapter for this if I can find how is the trick to generate +28 v for VGH. Otherwise I guess the FET gate won’t open to let pass 15v to the pixel and the whole effort can fail…

martinberlin, 2023-11-08:
Can someone please confirm the connector that needs to be on the adapter:
https://www.digikey.de/de/products/detail/panasonic-electric-works/AXE650124/2048713 sorry but I cannot see that photo correctly since it's out-of-focus.

ddB0515, 2023-11-08:
I can confirm AXE650124 is one you need, tested on my board with ES108FC1 
Closeup of AXE550127D from ES108FC1 FPC
[image]

Also as specified on Panasonic website for AXE550127D (check bottom The other side)
https://www3.panasonic.biz/ac/e/search_num/index.jsp?c=detail&part_no=AXE550127D

martinberlin, 2023-11-08:
Thanks!   Will take like 2 weeks but I've added 10 to JLC "My parts" using Global sourcing from Arrow

> verical #: AXE650124                                             JLCPCB Part #: C2924739 | 10 X $1.8548 + 1 USD handling fee

martinberlin, 2023-11-08:
@ddB0515 started with this if you can review the schematics it will be really nice to see if I got the pin-out correctly
https://github.com/vroland/epdiy-hardware/blob/40-to-50-axe-adapter/adapters/40-to-50-pin-adapter/40-to-AXE650124-schematics.pdf
Question: Anyone knows if we can completely ignore the pins 46 to 50 that are some kind of SPI external memory?
Because I could get only first 7 pages of the Datasheet.

ddB0515, 2023-11-08:
Posting here datasheet if there is issue with datasheet (not sure why only 7 pages you can see) 
[ES108FC1_ Ver0.1.pdf](https://github.com/vroland/epdiy/files/13300691/ES108FC1_.Ver0.1.pdf)

Also checked datasheet for pin 31
```
Note 5-2
Please connect to VDD voltage by 10K resistance.
```
and pin 33
```
Negative power supply source driver
```
Also for BORDER some say that should be connected to the GND and someone to VNEG not sure what is should be for this kind of screens :(

And yes everything (Except GND) from pin 35-50 can be NC)

martinberlin, 2023-11-09:
[image]
On the works if anyone detects any mistake in schematic or orientation of the output connector please let me know!

vroland, 2023-11-09:
Btw, can you break out the SPI flash pins as well? This could be quite useful for dumping the flash with the waveform. And if you rotate the AXE connector by 50 degrees it could be easier to route.

martinberlin, 2023-11-09:
> And if you rotate the AXE connector by 50 degrees it could be easier to route.

Though about that but then the display will also rotate 180 degrees which leaves it kind of hard to connect (I dislike to test expensive displays that hang over your table)
But will consider it, good point.

> Btw, can you break out the SPI flash pins as well?

Yeah sure will add then an optional 4 pin header with labels close to the input connector. Thanks for the feedback

martinberlin, 2023-11-10:
Routing passes DRC. If someone can take a look please go ahead:
https://github.com/vroland/epdiy-hardware/tree/40-to-50-axe-adapter/adapters/40-to-50-pin-adapter

I still need to wait like 10 or even 14 days till the 50 pin AXE connectors arrive to JLC so there is time. Short preview:
[image]
[image]
Not the best routing. So if you consider that you can make a better job, please make a pull request!

vroland, 2023-11-10:
Nice, looks pretty good to me. I'll just order some with my next order ;)

yamayama-mountain, 2023-12-12:
I found a display similar to ES108FC1, called ES108FC2. You can view it here: https://www.aliexpress.com/i/1005005413312620.html

I have also obtained the datasheet, which is attached.

Unlike ES108FC1, this display features a 0.3mm 51-pin flat cable connector, providing relatively better usability. 
Has anyone ordered this product before?

[ES108FC2)_ Ver 1.0 VB3300KHB.pdf](https://github.com/vroland/epdiy/files/13649483/ES108FC2._.Ver.1.0.VB3300KHB.pdf)

dengfengdz, 2023-12-25:
Hi ! guys.
I got a waveform about ES108FC1, but I didn't know how to convert it to data that epdiy could use.I don't understand the conversion format, can someone convert it to usable data.
Thank you!
[waveform.zip](https://github.com/vroland/epdiy/files/13764411/waveform.zip)

tomek-szczesny, 2024-01-18:
Just to keep this thread somewhat alive, I changed my plans and decided to build a different board.
It will have ICE40HX4K FPGA, driven by a single channel LVDS from Odroid M1S (so up to 1080p@30Hz should be possible).
Together with TPS651851, 16-level grayscale VGA output for reference, on-board oscillator for VGA, 64Mb PSRAM, debug buttons and LEDs.
Inspired by epdiy, I may add a few other connectors for other e-inks. Certainly a 6" model I have in a broken e-book reader.

I'm looking forward to collaborating with you guys to get ES108FC2 to work. I have already implemented blue noise dithering with monochromatic output on my other FPGA board, and it looks quite nice on 24" 1080p screen. I expect it to look even better on a smaller display.
I believe this dithering algorithm might have a chance of working on ESP32 as well, to allow very low response time.

tomek-szczesny, 2024-01-18:
Ah, also a word of caution: It looks like ES108FC2 slightly exceeds rating of TPS651851. Namely maximum current of -20V rail, which may reach almost 16mA, while TPS chip is rated at 15mA (or 12mA if you stick to TPS65185). 
This is a charge pump output, so exceeding current rating will not cause damage, just may cause temporary voltage drop. 

Since there are no obvious solutions to this problem, and there are virtually no alternatives for TPS chip, I'm willing to take that risk, but that's something I wouldn't let go in high volume production.

vroland, 2024-01-21:
I hope I can also try it soon, I already have the display and will have the connectors soon, AliExpress is shipping to Austria again starting this year :tada:

tomek-szczesny, 2024-01-21:
Holy moly, now I'm reconsidering my choices:
https://aliexpress.com/item/1005005791991415.html
[ED113TC1.pdf](https://github.com/vroland/epdiy/files/14003084/ED113TC1.pdf)

ddB0515, 2024-01-22:
@tomek-szczesny I have created new board with MAX17135 that is capable of -20V/20mA and should be perfect alternative for TPS65185
Also created dip switch for multi voltages 22/25/28 (Check image)
[image]
When I test all and confirm that is working will probably open source it (for now still in POC stage)

And yesterday I got mine ED113TC1 but need to make adapter for it
A bit short with time for doing sw implementation as just recently managed to get and source all parts but let's see...

tomek-szczesny, 2024-01-22:
@ddB0515 
Yeah I ruled out MAX because its availability is severely limited. Right now can't find any place to get it (and I'm not back-ordering anything from Mouser again). 

I've just designed today a voltage selector in 21-28V range in 1V steps, because I've seen 27V eink datasheet somewhere already.
It consists of 3x nfets adding parallel resistors to lower side of feedback divider, thus 8 combinations and 8 steps available. You could do the same with your switches, if you still have a chance to change your design. 

I'll probably just add the ED060 connector and adapter header. I gave up trying to guess what e-ink I might want to use in the future, and spent enough time contemplating it.

ddB0515, 2024-01-22:
@tomek-szczesny  sound like nice idea, if you already have tested solution with 3 nfets with resistors share here or send me message, more than happy to take a look :)

tomek-szczesny, 2024-01-22:
@ddB0515 I don't think there are private messages on GitHub..
My idea is just these notepad scribbles I've done bored at work today, so it's not tested. But knowing success rate of my designs I don't anticipate any troubles from this approach.

Switches will be MOSFETs in my design, I'm just too lazy to draw them properly.

I came up with a different value set that is available in E24 series that I stock at home, with 300k instead of 1M.
[image]

alexose, 2024-01-22:
Loving all the activity here.  To summarize:  You've both designed versions of the EPDiy that support a wider range of voltages?  That seems like a useful development towards supporting more screens.  Happy to help test the hardware when you think it's ready.

> I gave up trying to guess what e-ink I might want to use in the future

I think this is an important use case for EPDiy:  Future-proofing designs (within reason) so that later revisions of screen hardware will still work.

tomek-szczesny, 2024-01-22:
> Idea is nice and should work but ofc until is on board(or protoboard) not guarantee :)

Back in the previous industry I worked in, an engineering model (prototype) wasn't even considered before engineers proved their designs mathematically, 3-sigma reliability in all specified conditions and so on. That's why I haven't used my breadboard in years and just build things. :) I don't do formal analysis either, so screw ups still happen ;)

> To summarize: You've both designed versions of the EPDiy that support a wider range of voltages?

I didn't. I'm just here to learn about e-ink in general. :)
My design is FPGA based, and is meant to translate LVDS video to e-ink signals in real time.
My [current prototype](https://github.com/tomek-szczesny/m1s-fpga-playground) generates video test pattern internally, converts it to grayscale, applies blue noise dithering to monochrome, and displays that on VGA monitor in 1080p.
The next iteration will deal with actual LVDS signal and e-ink target, hopefully..

I think that designing and building adapters for future e-inks is inevitable, but it's easier and cheaper than building a new control board from scratch every time. This fact has clearly been recognized in epdiy already.

DerPicknicker, 2024-04-03:
Hey guys, 
how is the current status of the eink panels…? 

Is one of the mentioned boards are compatible with an esp32?

DerPicknicker, 2024-04-03:
@martinberlin … 

I saw that you published an adapter for the e-ink screens. While I wrote the entry above. I found a promising board: https://x.alibaba.com/Aw7qGg?ck=pdp

Would you say that your adapter will fit? This would be a great deal - 38$ for the driver which is well documented and also supported by several esp32 and raspberry pi projects.

martinberlin, 2024-04-04:
Hello @DerPicknicker 

> Would you say that your adapter will fit?

First of all this already has connectors so you don't need to use my adapter and those boards come pre-configured only to one size. Good-luck finding how to switch sizes via SPI with the poor documentation (most) chinese companies add in their products. Second: This is really epdiy Issue board where we should cope with epdiy related stuff and strictly to the Title thema, is not a discussion board, for that please join epdiy.slack.com (Send me an Email)
I would not go that way. Specially because you will need to add a second board to send the SPI framebuffer which is double consumption, double space, well I dislike the idea because I already did it and is not my thing.

DerPicknicker, 2024-04-04:
I will write you an email @martinberlin ...

martinberlin, 2024-04-04:
Use this link to join SLACK https://join.slack.com/t/epdiy/shared_invite/zt-2g3dpt45i-V5vleYuxcwoM9Rbz2tXtRA

martinberlin, 2024-04-07:
Sent 5x of this to fabrication: https://github.com/vroland/epdiy-hardware/tree/40-to-50-axe-adapter/adapters/40-to-50-pin-adapter
There are 2 versions of this adapter, for the moment just send 5 units of my take, and I still do not have the panel.

[image]

Linked to https://github.com/vroland/epdiy/discussions/290 since someone is hungry to try this